Ultimaker/Cura #8875: [4.8.0] crashes at startup when PyQt5 is built with QT_DEBUG

bnavigator, 2020-12-02:
As reported in https://bugzilla.opensuse.org/show_bug.cgi?id=1157504 (for an earlier version of cura)

**Application version**
4.8.0

**Platform**
openSUSE Tumbleweed Linux with distribution provided packages

**Reproduction steps**
1. Build and use PyQT5 with --debug added to `configure.py` or `sip-install/sip-build`
2. start cura

**Console output**
See also https://bugzilla.opensuse.org/show_bug.cgi?id=1179214
```
2020-12-02 17:20:25,421 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [350]: Post processing view created.
2020-12-02 17:20:25,498 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///usr/share/cura/resources/qml/PrintSetupSelector/Custom/MenuButton.qml:42:18: QML Label: Cannot anchor to an item that isn't a parent or sibling.
2020-12-02 17:20:25,522 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [100]: There was no auth data or access token
ASSERT: "PyDict_Check(py)" in file ../../qpy/QtCore/qpycore_qvariantmap.cpp, line 36
Fatal Python error: Aborted

Thread 0x00007fefce6db640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 79 in _updateThread
  File "/usr/lib64/python3.8/threading.py", line 870 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007fefedffb640 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 306 in wait
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 2432 in wait
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 1726 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007fefee7fc640 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 306 in wait
  File "/usr/lib64/python3.8/threading.py", line 558 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/usr/lib64/python3.8/threading.py", line 870 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007fefeeffd640 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 306 in wait
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 1476 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007fefef7fe640 (most recent call first):
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 1350 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007fefeffff640 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib64/python3.8/threading.py", line 870 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007ff009893640 (most recent call first):
  File "/usr/lib/python3.8/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/usr/lib64/python3.8/threading.py", line 870 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007ff00a094640 (most recent call first):
  File "/usr/lib/python3.8/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/usr/lib64/python3.8/threading.py", line 870 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007ff0267fc640 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 302 in wait
  File "/usr/lib64/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007ff026ffd640 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 302 in wait
  File "/usr/lib64/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007ff0277fe640 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 302 in wait
  File "/usr/lib64/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007ff027fff640 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 302 in wait
  File "/usr/lib64/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Current thread 0x00007ff06a718740 (most recent call first):
  File "/usr/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 302 in initializeEngine
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 912 in runWithGUI
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 829 in run
  File "/usr/bin/cura", line 235 in <module>
[1]    15921 IOT instruction (core dumped)  cura
```

Here is a diff of **one** offending code snippet, but there are more in the same file.

```diff
--- a/cura/API/Account.py
+++ b/cura/API/Account.py
@@ -258,7 +258,7 @@ class Account(QObject):
 
         user_profile = self._authorization_service.getUserProfile()
         if not user_profile:
-            return None
+            return {}
         return user_profile.__dict__
 ```

Ghostkeeper, 2020-12-03:
Hmm, yeah there are quite a few untyped functions there which have a `@pyqtProperty` that sets a type (e.g. `str`) but then sometimes return `None`. Not very nice of us! They don't show up in our CI since they are not typed with Python's type annotations for Mypy. But of course PyQt still restricts the type. I guess it works in release mode because it just assigns the QML property with `null` then which works through Javascript's weak typing.

Ghostkeeper, 2020-12-03:
I could see us fix these issues with boy-scouting. That's what I would do. But going through our entire code base to search for these issues is not something that will likely happen any time soon unless we have a need to go through the debug symbols of PyQt.

I thought @nallath had PyQt compiled in debug mode though. Or was that Qt? Or a different computer maybe?

GregValiant, 2024-11-29:
Thanks.  I'll go ahead and close this..

closed 2024-11-29 by GregValiant