Ultimaker/Cura #8876: Option to conditionally connect all raft elements

MarcinOrlowski, 2020-12-02:
I sometimes print projects that require raft. But in some cases raft alone is not enough and I'd like to add additional hooks by connecting separated rafts togeter. For example, this project: https://www.thingiverse.com/thing:571949 works better for me when I use raft. But slicing it with rafs enabled produces this (raft margin 6mm):

[image]

I could increase raft margin, but that wastes the print time and filament. This is with raft width doubled to 12mm:

[image]

So still last leg is not connected. This is a problem for me because sometimes (and whole print takes hours) the last left got loose and the whole print gets ruined. For now I just have small 0.4x0.4 mm rod SLT

[image]

 that I add to the project I print to trick Cura to produce connecting raft:

[image]

I believe I am not the only one so it would be good to have the option to solve the "raft problem" without the rod STL. For now I think that the sufficient sollution would be this: if "Connect raft elements" option is enabled then there's user defined radius for which standalone raft would be "bridget" to. That would make such raft fixed in place for good and solve the problem.

What you think?

konskarm, 2020-12-03:
Hi @MarcinOrlowski. We discussed this feature request today with the team and, although we see the value in it, we know that it won't get prioritized sometime soon and it will get lost in our already long backlog. Therefore we decided to defer it. The workaround you used with the thin model is the one I would personally suggest too for models like the one you posted.

MarcinOrlowski, 2020-12-03:
Thanks for considering the request. Quick question (I absolutely know nothing about Cura plugins, APIs and internals). Would that be doable by a plugin? I mean is plugin able to "pre-slice" to see in case raft is used and then alter geometry?

Ghostkeeper, 2020-12-04:
No, the slicing process is done in CuraEngine. CuraEngine has no plug-in structure (yet). Cura's plug-ins are only for the front-end.

What you could do from a plug-in is add a strip to the build plate similar to what your workaround is right now. That strip then gets sliced along with CuraEngine. That is very similar to what for instance the Support Blocker tool does (which is implemented as a plug-in too) or the Custom Support plug-in that you can find on the Marketplace.

closed 2023-07-13 by github-actions[bot]