Ultimaker/Cura #8878: Rezising cutom support unevenly ends up messing, changing the main object size

TheCuraMan, 2020-12-03:
**Application version**
4.8.0

**Platform**
Windows 8

**Printer**
Ender 3 Pro

**Reproduction steps**
1. 

Changing custom supports size unevenly , with uniform scaling unchecked after playing with size, rotation and positioning of customs supports I check the object size and it has changed.
Sometimes it changes in a very noticeable way because it changes the % number too, but many time if not paying special attention it would go unnoticeable because the % are all kept at 100% but the size numbers of one x, y or z has changed or two of them or all of them.
It is driving e nuts. First I did the changed I needed to the custom supports sliced and then opened the object again in another new cura session and compared sizes and they were different, the after many tries to try to understand what was going on and when exactly the size ended up messed up I decided to take snap shot o the object freshly imported and positioned on the bed, then again after working with the custom supports and slicing them each time I needed to check they were in the position I needed, when I do the final slice and compare the size of the object to the screenshot of the freshly imported and positioned one their sizes are different.
I even for the first time decided to lock the object positioning just in case to see if it would help but id didn't. Maybe a locking object size would be a good thing to add to cura.

[image]
[image]

After that last screenshot I just changed a little the custom support position and it changed the size of the object again.

And obviously I have being very, very careful to not change the object size by mistake.

[image]

konskarm, 2020-12-03:
I tried reproducin the problem with the given project file, but I do not see anything unusual. Regardless of how I move/rotate/resize the support objects, the model does not change size at all.

Here is what I tried:
1. Import the provided 3mf file
2. Ungroup the objects
3. Select the model and take a screenshot of the model's size
4. Select one of the custom supports
5. Uncheck uniform scaling
6. Resize/move/rotate the support object randomly
7. Check again the size of the model compaired to the screenshot

Result: The object's size is same as before.

Am I missing something? Could you provide a step-by-step reproduction scenario?

TheCuraMan, 2020-12-03:
> I tried reproducin the problem with the given project file, but I do not see anything unusual. Regardless of how I move/rotate/resize the support objects, the model does not change size at all. [...]

Hello.
Yes upload the original object: https://drive.google.com/file/d/13LYZAp4Qp2ZdLF5g573ranSOdbYcHab2/view?usp=sharing

then position it flat on the bed, select the object with the resize tool to see it's size and take a screenshot of the numbers.
Then just add custom supports to improve the base of the supports suggested by cura.
I choose 1.2 layer height on the project so support suggested by cura are from 59% and up.
Or you can just try to match my screen shoot. But I sliced the model first to see the proposed supports then try to add a custom support for one of them slice to see if the location was ok, adjust the orientation, size and position of the custom support , slice, then move to another one, etc
That was the way I did it.
One of the reason I did this is because this huge object does not need brim for adhesion but sadly cura does not have an option to get brim only on supports if the object does not have support, that option could be added. Also It would be grate to be able to choose more layers for the support Brims. Sometimes there are supports that are very tiny and delicate and it would be great to be able to have a Brim that is 3 layer high for example.

I tried the Tower support, to beef up  some of them but sadly it affect all the supports and I didn't want the supports that are rooted on the object body to get beefy, only the ones on the bed plate. Maybe that would be another great option, to be able to use tower support only on the ones that are on the beds plate. 
 
I have had this randomly main object resized by cura problem many, many times before with many models and is super frustrating.

I thought that if I unchecked the uniform scaling with the custom support selected, resized it then immediately check back the uniform scaling I could avoid cura altering the object size but this this time I  realized that although the object was showing 100% on x, y and Z the actual decimals number of one of them was altered, changed.

At the moment it seems it only comes up when the uniform box is unchecked and custom supports are resized like that. I am not sure if rotating the custom support is part of the equation too, but do it also.

I hope you can replicate it on your end so the issue could be fixed. I have lost many hours on different models because of this.

If you are not printing stuff that needs to fit in to one another it probably does not matter because usually is just the decimals that change.

But when you print things that need to fit or human bodies that need to have good proportion then it can be noticed on the print.

Ghostkeeper, 2020-12-04:
I'm not getting the same results here.

Here is a screenshot of the object size when I first load the model and lay it flat on the build plate:
[image]

And then I added two cubes, turned them into support meshes and repositioned them in similar spots to what you showed.
[image]

One thing that might be happening is that the Custom Support plug-in on the marketplace adds the Support Meshes cubes as a sub-mesh of the mesh you're clicking on. They are seen as part of the mesh then (and scale and rotate along if you rotate/scale the parent mesh). As such they are also counted towards the total size of the mesh, but don't actually change the size of the total mesh. I didn't use that plug-in here so maybe that's why I couldn't reproduce it. Did you use a plug-in to create the Custom Support?

The total size includes the cubes but the percentage scale also changes. The reason for this is that the model has also been rotated (due to laying it flat on the build plate) and so the X, Y and Z dimensions changed relative to its original size. Cura compensates for this by computing the total rotation and doing an Euler transformation on the X, Y and Z dimensions to get at the original size. However because the object was then made larger by the support meshes I imagine it doesn't take that into account.

For other problems, please create separate tickets. It would make this discussion a lot harder to follow to discuss them all here.

> I have had this randomly main object resized by cura problem many, many times before with many models and is super frustrating.

Note that we had a bugfix in 4.8 to fix this issue: https://github.com/Ultimaker/Cura/issues/4136 So some of the issues you encountered before might not occur any more now.

TheCuraMan, 2020-12-04:
Well In your last picture I see you have uniform scaling checked and although I see you resized one of the custom supports unevenly I am not sure why you got it checked again. Also you have snap scaling checked and I have never checked that, I am not ever sure what it does.
Also I have only come aware of this issue on cura 4.8.0 I don't remember if I ever got this problem on 4.7.x 
I just recorded a video of me doing this so you can check it, and of course I got same results as other times. https://drive.google.com/file/d/1k7_OaMFRlnPV9PGPaF2d2vxIcNZiz99z/view?usp=sharing

[image]

[image]

I understand your explanation I would think maybe is something related with that but to check it I deleted the supports and checked the object size after and although the 110% came down to 100% the decimals were kept a little different.
What I know its that is quite nerve raking to see the object size change when we are preparing to print object that can take a day or more and quite a lot of filament.
Also is even more nerve raking when we want to print a model that has many parts and instead of printing everything at the original size we want to print it at 70% or 180% for example. To see the objects sizes changing randomly with out understanding what is going on and what the final outcome will be it is stressful, at the end many hours and material could all be lost if thing don't fit properly or if anatomic proportions end up out of place.

Ghostkeeper, 2020-12-07:
> Well In your last picture I see you have uniform scaling checked and although I see you resized one of the custom supports unevenly I am not sure why you got it checked again. Also you have snap scaling checked and I have never checked that, I am not even sure what it does.

That's just because I was using Shift+Ctrl+PrintScrn to create my screenshot, sorry. Shift toggles Snap Scaling. Ctrl toggles Uniform Scaling. They were not actually used to scale the objects.
Snap scaling snaps the scale factor to increments of 10% when dragging the scale handles.

> I just recorded a video of me doing this so you can check it, and of course I got same results as other times. https://drive.google.com/file/d/1k7_OaMFRlnPV9PGPaF2d2vxIcNZiz99z/view?usp=sharing

Ah, so you weren't using the Custom Support plug-in. You were using the Support Blocker tool as an easy way to create cubes on the build plate (rather than having to load cube.stl). And then changed those cubes into support meshes rather than support overhang blockers. **It would've been useful to write that in your reproduction steps!**

Those support blockers are added as being part of the main mesh. Normally support blockers, cutting meshes and infill meshes don't get counted towards the size. But support meshes do, because those are actually printed. So it's kind of expected that they are counted towards the size of the mesh. But it feels weird since you've only selected the parent mesh, not all of the child meshes too.