Ultimaker/Cura #8879: Tree Support in Cura 4.8.0 is buggy

BERT-LZ, 2020-12-03:
Tree Support in Cura 4.8.0 is buggy: Distance to the bevel is much too big, so it makes no sense at all:
 
[image]

If you switch to normal instead of tree support, the set distance will fit:
 
[image]

With the same settings with Cura 4.5.0 it still worked correctly:
[image]

BERT-LZ, 2020-12-03:
[treesupport-testfile.zip](https://github.com/Ultimaker/Cura/files/5635482/treesupport-testfile.zip)

BERT-LZ, 2020-12-03:
This gap remains if Z and X/Y distance is set to 0 for test purposes. Why does this gap remain? There is a huge bug here, because it makes the tree support almost unusable!

[image]

Asterchades, 2020-12-03:
Turn off the Support Interface - at least on the top. It's already well known that 4.7 and beyond trees and the interface don't work well together when it comes to horizontal (X/Y) proximity.

Ghostkeeper, 2020-12-04:
Duplicate of https://github.com/Ultimaker/Cura/issues/8622.

This didn't occur yet in 4.7 as far as we know. It's new to 4.8.

closed 2020-12-04 by Ghostkeeper