Ultimaker/Cura #8881: generated lithophane model cannot be selected!!

icefox1983, 2020-12-04:
**Application version**
(The version of the application this issue occurs with.)
4.8.0
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Mac
**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
open Cura. Drag a jpg file onto the platform. Choose any setting. A model is generated. The model cannot be selected, thus cannot be moved or scaled. The problem is reproduced every time. Makes the version unusable!! Please fix!!!
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Vandrasc, 2020-12-04:
Hi @icefox1983 , thank you for your report, though there is not much we can do without additional information. Please feel in the bug template so we can help you as fast as we can. Thank you!

icefox1983, 2020-12-04:
Can u try reproducing it first? I didn’t include all details because the bug is obviously the software problem and anyone with a Mac can reproduce it

Thanks,
Bin

Vandrasc, 2020-12-04:
I did, unfortunately, I am not able to reproduce, my model can be selected each time. 

_"I didn’t include all details because the bug is obviously the software problem and anyone with a Mac can reproduce it"_  --- I do get your point but as you can see not everybody with a mac can reproduce the issue, this also being the reason we ask for the bug template, helps us understand the issue better. 

Let's try different, what os x version are you using? Is it always reproducible, or only on a certain image? Any logs, files you can share with us?

icefox1983, 2020-12-04:
Wow that’s unexpected. In this case I’ll provide more info. For now, I’m on the latest Big Sur version, intel iMac 27”, with a magic touchpad, and yes the problem persists for me every time for different pictures and different generated mode sizes. That’s why I thought it’s reproducible for everyone. In fact I’ve seen this issue before, the really thin/tiny models would sometimes be not selectable, although reloading them general solves the problem. This time it never works. 

Thanks,
Bin

Vandrasc, 2020-12-04:
Oke, so this is a common issue for users who upgraded to Big Sur, see also https://github.com/Ultimaker/Cura/issues/8745 and https://github.com/Ultimaker/Cura/issues/8831. 

Good thing is that we already have it on our backlog and hopefully we can address it soon. 

Thank you for your feedback!

icefox1983, 2020-12-04:
Ha so it is reproducible, although just for Big Sur users. That’s what I thought. Thank you. 

Thanks,
Bin

fieldOfView, 2020-12-04:
And that is why we ask for the platform, and why it is very helpful if you say more than just "mac". Yes, details are important. If they weren't, we would not ask for them.

Ghostkeeper, 2020-12-07:
Closing as duplicate of https://github.com/Ultimaker/Cura/issues/8745.

It'll get triggered because of the pop-up that loading images produces.

closed 2020-12-07 by Ghostkeeper