Ultimaker/Cura #8882: Nozzle Size dropdown for Custom Printer shows 'EMPTY'

RFBomb, 2020-12-04:
**Application version**
4.8.0

**Platform**
Windows 10, 64 bit

**Printer**
Creating a custom printer. Files uploaded here:
[Custom Printer Resources.zip](https://github.com/Ultimaker/Cura/files/5640617/Custom.Printer.Resources.zip)

**Reproduction steps**
1. Essentially I took the Ender 3's .def.json file, gave it a new name, copied in the settings I plan to fine tune, and added a second extruder to the printer definition. 

2. Then, I took the creality base extruder file, copied that into 2 new files to be used for the new custom printer. 

Updating the settings (such as the extruder names) in the extruder files will cause a newly added printer to have the new names, so I know that it is reading in the files properly. Cura is allowing me to add the printer in and do customization, but will not allow selection of a nozzle size. 

For reference, the 'ender 3' stock profile works correctly. I even tested a direct copy-paste of that file in my new filename, and it had the same issue. 

**Screenshot(s)**

Ender 3:
[image]

Custom Ender 3 Profile:
[image]

**Actual results**
See pictures above

**Expected results**
the custom profile should have same nozzle selection as the ender 3. 

**Project file**
[Custom Printer Resources.zip](https://github.com/Ultimaker/Cura/files/5640617/Custom.Printer.Resources.zip)

To test the custom printer profile yourself:
1- Unzip the resources file I uploaded. It contains the following:
     \resources\definitions\custom_Ender3_MicroSwiss_DirectDrive.def.json
     \resources\extruders\MicroSwiss_DirectDrive_extruder_0.def.json
     \resources\extruders\MicroSwiss_DirectDrive_extruder_1.def.json

2 - Drop the 'resources' folder into your install folder. Mine was located here: C:\Program Files\Ultimaker Cura 4.8.0\
      This should drop the files where they belong. 

3 - Open cura (Restart if already open). 
4 - Add new printer. Printer will be found in the 'Custom' section. 
[image]

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5640652/cura.log)

Ghostkeeper, 2020-12-04:
Nozzle profiles are specific to the printer too, and would need to be copied as well. You can find the Ender 3's nozzle profiles in the `variants` folder. In those files you'll need to adjust the printer they apply to.

To have a complete copy of the Ender 3 you would then also need to look through the materials. `generic_petg_175`, `generic_abs_175` and `generic_pla_175` also specialise settings for Creality printers, so that would need to be applied to your printer as well. They change some important parameters such as the printing temperature and the build plate temperature.

closed 2020-12-04 by Ghostkeeper

RFBomb, 2020-12-04:
So I have to create a new set of nozzle files in the ‘variant’ folder? Is
that ‘best practice’ here? Or can they Use/inherit currently existing ender
3 nozzle files be referenced (it will be using the same nozzle specs).

As far as material specs go, those are set up printer specific are they?

Ghostkeeper, 2020-12-07:
> Or can they Use/inherit currently existing ender
3 nozzle files be referenced (it will be using the same nozzle specs).

Yeah, you'll have to actually create new nozzle profiles. There's no way around that now. We didn't implement anything to reference other printers' nozzles since the amount of nozzle profiles currently isn't very significant as to impact performance in any meaningful way, and it's not an issue with maintenance since it's rare that the nozzle profiles need to be exactly the same in multiple printers. UM3 vs UM3 Extended is the only example in Ultimaker's ecosystem, and Ultimaker's product line is quite varied as FDM printer manufacturers go.

RFBomb, 2021-01-04:
@Ghostkeeper 
I'm commenting here since the question is related and I don't think it is worth opening up a new issue for my followup question. 

Before I get to the question, thanks for your help, I am now able to print successfully using the custom printer files. Nearly everything is working as expected. The issue I have though is that I would like to specify a generic Retraction distance and speed that overrides the default settings if the material itself does not have the settings specified. 

For example: If I switch between Generic 175 PLA and Generic 175 PETG, the retraction distance defaults to 6.5mm since I have not specified it in either material file for my printer. My hope was that by specifying `"retraction_amount": { "default_value": 1.0 },` in the Extruder.json overrides section that it would default to 1.0mm, but this is not the case. 

The standard extruder overrides settings (nozzle offset for example) are all working properly, but anything related to print settings has no effect. I'm assuming this is either intentionally ignored (or just never programmed due to being unexpected), or overridden by `fdmprinter.def.json` (which is where the 6.5mm default comes from). I'm aware that I can set it up in my `Printer.Def.json` file, but was hoping to have it able to be set per extruder (since in this setup 1 will be bowden and 1 will be direct drive).

Ghostkeeper, 2021-01-14:
It's a bit hard for me to see what's wrong precisely. One common mistake is that `"value"` always overrides `"default_value"`, even if your definition overrides `"default_value"` but the parent definition overrides `"value"` (documented [here](https://github.com/Ultimaker/Cura/blob/master/docs/profiles/getting_a_setting_value.md#evaluating-a-definition)). However in the case of `"retraction_amount"` fdmprinter at least doesn't override `"value"`, so that's not the problem you're facing unless you also wrote a `"value"` yourself. And then you'd know.

A tool that I like to use to trace and debug this sort of problem is the [God Mode](https://github.com/sedwards2009/cura-god-mode-plugin) plug-in. It adds a menu entry to the Extensions menu that allows you to show all setting overrides on the active stack. Using that tool you can see how Cura comes to a certain setting value. So you'll see, for instance, that the definition does set the retraction amount to 1.0 but then the quality profile overrides that to 6.5 or something.

RFBomb, 2021-01-14:
Interesting.
I’ll try that out and see what happens. But for now, I moved the value into
the printer setup file and it got what I wanted. Since the 2 hot ends for
my mod need to be the same (dimensionally), I think having it set as 1
value that applies to both will work just fine.