Ultimaker/Cura #8883: 4.8.0 Object not dropping to build plate after rotating

robotikosmokykla-lt, 2020-12-04:
Application version
4.8

Platform
win10

Printer
Ultimaker 2+

Reproduction steps
After updating to 4.8 from 4.7, and opening a .stl made in Solidworks, and rotating the object outside the build platform, 1/3 of it remains under the build platform. Moving to the platform does not change height. 

Slices well and prints it, starting from 2/3.

**Screenshot(s)**
[image]

**Actual results**
Prints half an object.

**Expected results**
A printer that prints all the object.

**Log file**
[cura - Copy.zip](https://github.com/Ultimaker/Cura/files/5643102/cura.-.Copy.zip)

Project File
[UM2_duru-rankena-pt2 (2).zip](https://github.com/Ultimaker/Cura/files/5643088/UM2_duru-rankena-pt2.2.zip)

rburema, 2020-12-08:
If you have 'drop to buildplate' on, a workaround is to drag the object up & let it 'fall'.

rburema, 2020-12-08:
In the picture, the gantry is very low (the blue lined box). Do you have one-at-a-time mode on or altered the height of the printer?

Objects may not land on the build-plate if they're above the height of the gantry! (This would mean they would be shown as 'striped' though.)

Ghostkeeper, 2020-12-08:
The expected behaviour is that if your object starts X mm below the build plate before rotation, it will still be X mm below the build plate after rotation.

As far as I can see that's what is happening here. The model in your project file starts at Z-9.5, and after rotation is still 9.5mm below the build plate. If the model starts on top of the build plate, it also ends up on top after rotation.

robotikosmokykla-lt, 2020-12-08:
This is a serious bug that affects reliability in that many hours of
manufacturing can be lost this way. If the setting is "drop to buildplate"
expected behavior is that all models will be printed from 0 not minus Z.

Ghostkeeper, 2020-12-08:
Well at some point you moved your model to minus Z. It stays there unless told otherwise.

robotikosmokykla-lt, 2020-12-08:
It was told otherwise when I brought it to the buildplate. At no point it
was moved to minus Z by me. It was rotated and probably that way it got
down to minus. It was it's own logic that if while rotating it gets under,
it stays that way. The interesting part is that rotating withing the
buildplate it gets above buildplate as expected.

robotikosmokykla-lt, 2020-12-08:
Tested on 4,5 cura. Same behaviour. Then it seems this is a featurebug. I
still do not understand why I should consider model behaviour being
different outside the platform than inside and especially why that does not
change while bringing the model back. So if not a bug, this is an UX
problem. IN that case the issue can be changed to a suggestion to
standardize behaviour and make the "drop to buildplate" option actually do
that.

Ghostkeeper, 2020-12-09:
> At no point it was moved to minus Z by me. It was rotated and probably that way it got down to minus.

Well that would be a bug then. However I can't reproduce it :( It stays on the same Z coordinate no matter how much I rotate the model. Do you remember the reproduction steps that caused it to be moved to a negative Z coordinate in the first place?

Models that go outside of the build volume are not aligned with Z=0 since it causes people to lose big models that need to be scaled down if they are so big that they go up and over the viewport.

robotikosmokykla-lt, 2020-12-09:
Got it recorded how I see the problem and the settings used:
https://vimeo.com/488914931/e06669703c

Ghostkeeper, 2020-12-18:
Ah, I see what happens now. Thanks for the recording. And you've also described it before, but I glossed over it then. All platform physics are explicitly disabled if the model goes out of the build volume.

So while the model is outside of the build plate, you can indeed rotate it such that part of the model is below the build plate. This is done because people use the space outside of the build volume to keep items separate that they don't want to print, in order to more easily arrange multiple print jobs. It's considered to be annoying that their objects move around there. Collision avoidance is also disabled there for the same reason.

I kind of understand where you're coming from. Either way we break some workflows though. And the workflow of moving something out of the build volume, rotating it there and then moving it back is not something I'd consider common. Why would you move it out there first? And I'd say that it's pretty obvious that some part will be missing. It's rendered as a different colour in the prepare stage, and missing in the preview stage. Such things are hard to judge for me though.

Ghostkeeper, 2020-12-18:
This is currently expected behaviour so I'll remove the "bug" label. Any other opinions on whether or not platform physics should be executed for objects outside of the build volume?

robotikosmokykla-lt, 2020-12-18:
Thank you, I think I'll have it in mind now but might forget in the future
and get a failed part in the morning again. To my mind if the setting is
drop to buildplate, then Cura shoul always drop to buildplate no matter
from Z+ or Z-. Now it get's a bit tricky that there is this hidden
exception that if you modify outside the plate, then this setting stops
working.

The need to rotate outside the build plate arises when preparing a larger
batch, after multiplying some parts move outside, then I start thinking how
to position them back into the plate.

Regards
Paulius

GregValiant, 2026-09-10:
I'm going to close this.

closed 2026-09-10 by GregValiant