Ultimaker/Cura #8886: Change at Z and Fan Speed for "Target Layer Only"

GregValiant, 2020-12-04:
**Application version**
4.8

**Platform**
Win 10 Home

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Enabled Change At Z to alter the Fan Speed and Print Speed for a "Target Layer Only".
2. Sliced the model and saved to file
3. Opened the resulting Gcode file in Notepad.

**Screenshot(s)**
N/A

**Actual results**
Fan speed and Print Speed were altered at the start of the Target Layer.  The fan speed was not returned to default at the end of the Target Layer
;LAYER:0
M106 S76.5    ;Starting Fan Speed

;LAYER:2
;[CAZD:
M106 S255
M220 S80
;:CAZD]

;LAYER:3
;[CAZD:
M220 S100
;:CAZD]

**Expected results**
The fan speed should have returned to the previous speed of 30%
;LAYER:3
;[CAZD:
M106 S76
M220 S80
;:CAZD]

**Project file**
[25x25Cube.zip](https://github.com/Ultimaker/Cura/files/5645029/25x25Cube.zip)
**Log file** In the Zip file

**Additional information**
When changing _only_ the fan speed for "Target Layer Only" the speed is changed for the Target Layer and at the start of the next layer it appears that a line is inserted with only the New Line character.  There are no CAZD lines.
;LAYER:2
;[CAZD:
M106 S255
;:CAZD]

;LAYER:3
                                                ; An empty string??  With Change at Z disabled - this is not a blank line
G0 X105.926 Y110.766 Z1

Additional:
When attempting to change the fan speed, and within the normal Cura settings the fan is starting slow and working up layer by layer to the set speed, Change At Z Fan Speed inserts the new speed right before the Cura calculated speed.  The result of that is to negate the Change At Z Fan Speed change.
;LAYER:4
;[CAZD:
M106 S255
;:CAZD]
M106 S33.7
;MESH:25x25Cube.stl

rburema, 2020-12-08:
Thanks for the in-depth look! We've decided to merge this with internal ticket (on backlog) CURA-7819 as the underlying cause is suspected to be the same as #8574 . Keeping this open, so we can reference it in case it's something different after all.

GregValiant, 2020-12-08:
I think there are a couple of issues.  In the case of this particular problem (not returning the fan speed at the end of the layer) you may just have a typo in the code so all that is coming in is the new line character.
The other issue is the problem of determining the Printing Z Height in a file with Z-hops, and changing the fan speed right after ";layer:xx" but before the automatic fan speed increase (because a user chooses to raise the fan speed every layer until it hits the target speed and then tried to use ChangeAtZ in conjunction with that).  If the fan speed change dictated by ChangeAtZ was inserted 5 lines after ";layer:xx" it would likely work since in the great majority of cases it would be after the automatic speed increase line.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-24 by github-actions[bot]