Ultimaker/Cura #8887: There is no retraction

viper0078, 2020-12-05:
I checked the GCODE because the filament is badly threaded and there is no retraction code.

[image]
[image]

[CFFFP_test.zip](https://github.com/Ultimaker/Cura/files/5646455/CFFFP_test.zip)

Is this a bug?

eskeyaar, 2020-12-05:
you can increase the  maximum retraction count to 10 and see ? That should make retract for you.

viper0078, 2020-12-05:
Thanks for the reply.

I see that the retraction was suppressed by the combination of MuxmumRetractionCount and MinimumExtrusionDistanceWindow.
I had two retractions at a distance of 1-2 mm with the extruder and I had to adjust the value because I had cut the filament, but I didn't think it would have such a negative effect.
I have the impression that Cura can make advanced settings, but it is difficult to print consistently.

closed 2020-12-05 by viper0078