Ultimaker/Cura #8889: Can not set decimals in steps / mm

Denethor85, 2020-12-06:
In Cura 4.8 can not set decimals in steps / mm under "Printer Settings" tabs.

rburema, 2020-12-08:
We've had some problems like this in the past. What are your locale/language settings (in both the OS and Cura)?

Does it work if you use a dot (`.`) as a decimal separator instead?

fieldOfView, 2020-12-08:
This is a different issue. The steps per millimeter settings are defined as int, not float. So they are specifically set to not have decimals.

Denethor85, 2020-12-08:
My locale Is it-IT, i use the italian language in Windows 10 and Cura 4.8. 
When I try to change the steps / mm in "Printer Settings" does not allow me to use any decimal separator, neither the comma nor the point, so consequently it only makes me inserts and integers.

Denethor85, 2020-12-08:
> This is a different issue. The steps per millimeter settings are defined as int, not float. So they are specifically set to not have decimals.

Yeah, but I don't think it's a desired thing? and if so I think it's an error because often the step / mm have decimals.

P.s. Sorry for my English

fieldOfView, 2020-12-08:
An aside question: what do you expect to gain from specifying the steps per mm in Cura? I did a quick search in the CuraEngine sourcecode, and it does not look like the steps_per_mm settings are used at all.

Denethor85, 2020-12-08:
> An aside question: what do you expect to gain from specifying the steps per mm in Cura? I did a quick search in the CuraEngine sourcecode, and it does not look like the steps_per_mm settings are used at all.
> 
> Edit: ah, I see, they are used in the X3G Writer plugin. Unless you use the X3G Writer plugin, you can ignore those settings.
> @Ghostkeeper, should/could these settings actually be float?

Great thing to know. I didn't know if those values ​​were used or not. As a novice, I thought of setting these values ​​ad hoc for my printer. I thought they were used for the calculation of movements and times

Ghostkeeper, 2020-12-08:
G-code (Cura's output) specifies distance in millimetres, not in numbers of steps. The conversion is done in firmware if that firmware accepts g-code.

It was added to Cura with the idea that it might be useful to snap positions to steps or certain microsteps depending on the desired print quality. You can reach greater print speeds by not using microsteps. Of course that was never implemented, but it doesn't do a lot of harm to have these settings there, being hidden by default. And it's used by the X3GWriter because the X3G format writes steps, as I understand.

The setting could be floating point indeed, at least in Marlin: https://github.com/MarlinFirmware/Marlin/blob/8e1ea6a2fa1b90a58b4257eec9fbc2923adda680/Marlin/src/module/settings.cpp#L165

commit bfe62514 (2020-12-08): Allow steps per mm to be float

You could have 333.333 steps per mm, for instance. Not common, but it's possible.

Fixes #8889.
closed 2020-12-08 by Ghostkeeper