Ultimaker/Cura #8891: cura 4.8.0 crash when loading .3mf file with metadata element

kcris, 2020-12-06:
Attached below the same 3mf file, exported twice
* without cura metadata (mini_bridge.3mf)
* with cura metadata (mini_bridge.cura.3mf)

The only difference between these 2 3mf files is the following section
```
[cura-crash.zip](https://github.com/Ultimaker/Cura/files/5649415/cura-crash.zip)
			<metadatagroup>
				<metadata name="cura:extruder_nr" type="string" preserve="1">1</metadata>
				<metadata name="cura:infill_mesh" type="string" preserve="1">True</metadata>
			</metadatagroup>
```
The 3mf file without cura metadata is loaded in cura 4.8.0.
The second one results in a cura crash.

nallath, 2020-12-07:
I can't reproduce this. Could you also share your configuration folder? It might be that the active machine causes this issue.

kcris, 2020-12-07:
sure, here it's the contents of my `AppData\Roaming\cura\4.8` folder, zipped
[cura4.8-configfolder.zip](https://github.com/Ultimaker/Cura/files/5653392/cura4.8-configfolder.zip)

nallath, 2020-12-07:
I've found (and fixed the issue). 

As a work around: Add an extra extruder to your custom ff printer. The file that you had referenced an extruder that was not available.

closed 2020-12-07 by nallath

kcris, 2020-12-07:
Thanks, this helps a lot! 
I will make sure to check and use the appropriate extruder!

commit 7096f64c (2020-12-07): Fix crash when extruder could not be found

Fixes #8891