Ultimaker/Cura #8892: Cura 4.8 wont load

Devtech223, 2020-12-07:
**Application version**
4.8

**Platform**
Windows 10 Nvida 2060Super GPU

**Printer**
Ender 3V2

**Reproduction steps**
won't load gives me only a blank white screen and stops responding in task manager

**Screenshot(s)**
[image]

**Actual results**

**Expected results**

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
2020-12-06 21:27:30,319 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-12-06 21:27:30,320 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-12-06 21:27:30,322 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2020-12-06 21:27:30,352 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-12-06 21:27:30,369 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-12-06 21:27:30,385 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-12-06 21:27:30,402 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-12-06 21:27:30,419 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-12-06 21:27:30,452 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2020-12-06 21:27:30,470 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2020-12-06 21:27:30,492 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2020-12-06 21:27:30,509 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-12-06 21:27:30,525 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2020-12-06 21:27:30,546 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2020-12-06 21:27:30,562 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
2020-12-06 21:27:30,579 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ModelChecker
2020-12-06 21:27:30,596 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MonitorStage
2020-12-06 21:27:30,617 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PerObjectSettingsTool
2020-12-06 21:27:30,633 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PostProcessingPlugin
2020-12-06 21:27:30,650 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-12-06 21:27:30,666 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PreviewStage
2020-12-06 21:27:30,691 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RemovableDriveOutputDevice
2020-12-06 21:27:30,706 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-12-06 21:27:30,738 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimulationView
2020-12-06 21:27:30,753 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SliceInfoPlugin
2020-12-06 21:27:30,769 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-12-06 21:27:30,786 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-12-06 21:27:30,871 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-12-06 21:27:30,872 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-12-06 21:27:30,874 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-12-06 21:27:30,876 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-12-06 21:27:30,877 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-12-06 21:27:30,879 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-12-06 21:27:30,880 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-12-06 21:27:30,882 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-12-06 21:27:30,887 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin Toolbox
2020-12-06 21:27:30,904 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CameraTool
2020-12-06 21:27:30,926 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MirrorTool
2020-12-06 21:27:30,948 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RotateTool
2020-12-06 21:27:30,971 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ScaleTool
2020-12-06 21:27:30,986 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SelectionTool
2020-12-06 21:27:31,007 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TranslateTool
2020-12-06 21:27:31,022 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TrimeshReader
2020-12-06 21:27:31,043 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPReader
2020-12-06 21:27:31,061 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPWriter
2020-12-06 21:27:31,082 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UltimakerMachineActions
2020-12-06 21:27:31,363 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UM3NetworkPrinting
[... 273 lines cut ...]
2020-12-06 21:34:08,063 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2020-12-06 21:34:08,065 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 457.51
2020-12-06 21:34:08,066 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
2020-12-06 21:34:08,068 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: GeForce RTX 2060 SUPER/PCIe/SSE2
2020-12-06 21:34:08,070 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
2020-12-06 21:34:08,072 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\default.shader]...
2020-12-06 21:34:08,076 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\selection.shader]...
2020-12-06 21:34:08,078 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\select_face.shader]...
2020-12-06 21:34:08,081 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\default.shader]...
2020-12-06 21:34:08,084 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\composite.shader]...
2020-12-06 21:34:08,087 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\overhang.shader]...
2020-12-06 21:34:08,091 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\striped.shader]...
2020-12-06 21:34:08,095 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\transparent_object.shader]...
2020-12-06 21:34:08,098 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\striped.shader]...
2020-12-06 21:34:08,101 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray.shader]...
2020-12-06 21:34:08,104 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray_composite.shader]...
2020-12-06 21:34:08,109 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\default.shader]...
2020-12-06 21:34:08,112 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\grid.shader]...
2020-12-06 21:34:08,116 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\platform.shader]...
2020-12-06 21:34:08,125 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray.shader]...
2020-12-06 21:34:08,631 - DEBUG - [Thread-3] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_cr10 took 0.40242934226989746 seconds.
2020-12-06 21:34:08,633 - DEBUG - [Thread-3] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-12-08:
Short answer: Please reinstall graphics card drivers.

Longer explanation:
I think I can see the problem:
~~~
2020-12-06 21:34:08,063 - DEBUG - [MainThread] UM.View.GL.OpenGL.init [112]: Initialized OpenGL subsystems.
2020-12-06 21:34:08,065 - DEBUG - [MainThread] UM.View.GL.OpenGL.init [113]: OpenGL Version: 4.1.0 NVIDIA 457.51
2020-12-06 21:34:08,066 - DEBUG - [MainThread] UM.View.GL.OpenGL.init [114]: OpenGL Vendor: NVIDIA Corporation
2020-12-06 21:34:08,068 - DEBUG - [MainThread] UM.View.GL.OpenGL.init [115]: OpenGL Renderer: GeForce RTX 2060 SUPER/PCIe/SSE2
2020-12-06 21:34:08,070 - DEBUG - [MainThread] UM.View.GL.OpenGL.init [116]: GLSL Version: 4.0.0
~~~
As you can see, the OpenGL version is 4.1.0, while the GLSL version is 400. This should never happen. OpenGL version 4.1.0 is locked to GLSL 410. I've seen this in the past, and the re-installation of graphics cards drivers fixed the problem for at least one reporter (maybe more).

Devtech223, 2020-12-09:
Crash on startup : White screen #8904
my issue was the same as this one and his known fix fixed mine and a graphics card reinstall did nothing to fix it

rburema, 2020-12-09:
@Devtech223 I hadn't seen that one, thanks for pointing it out! Are your 'reproduction steps' the same as @MooM00 s?

I'm going to close the newer issue #8904 as a duplicate (keeping this one open so we can track it), but I think it's useful to at least put the workaround found in the other issue here so people don't have to play 'follow the link':

> Set 'window_left' to any positive value in [general] section of %appdata%\cura\4.8\cura.cfg

Reproduction steps seem to have something to do with having 2 monitors. I use 2 monitors as well, but the primary one is on the right at the moment.

Devtech223, 2020-12-13:
@rburema Yes my repoduction steps where the exact same as @MooMOO
And I can confirm my primary monitor is my left monitor

fieldOfView, 2020-12-13:
With multiple monitors, try temporarily disabling one. If Cura starts with only a single display, then go to the General preferences in Cura and deselect the option to "Restore window position on start". Then you can reenable the second display.

alex20465, 2020-12-13:
Cura 4.7.1:

[image]

Possibly not relevant but there are a lot of symptoms to deal with in cura. Root cause no software testing, no QA.

fieldOfView, 2020-12-13:
> Root cause no software testing, no QA.

Well, you are great at motivational speaking, so you have that going for you.

> Possibly not relevant

Feel free to participate in issues, but please try to chime in where it *is* relevant. Yours is a very different problem than what is being discussed in this issue. See https://github.com/Ultimaker/Cura/issues/8472

mrx23dot, 2020-12-19:
Also crashes on Intel HD 2500 with OpenGL 3.1.
cura 4.7 at least works on it (although without simulation)

Please keep compatibility with old HW, there is no point buying new PC only for slicing. (I wouldn't call displaying a 3D model as cutting edge application), let's think green.

Also it would be time to do CPU/memory profiling, it's getting slower and slower even on my 4core desktop, and eating 500MB RAM, >20-30sec startup.

nallath, 2020-12-21:
We know what things are supported by various OpenGL versions, there no need to tell us again. We also regularly profile Cura. 

But as others have already posted before; Without adding extra information about your system, saying that the issue occurs with you is simply not helping.

mrx23dot, 2020-12-21:
I just included what OpenGL features are supported by my older Intel card, so simulation could be fixed for it.

Please find my 4.8 crash logs attached:
[stderr.log](https://github.com/Ultimaker/Cura/files/5724773/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/5724776/cura.log)
[image]

Ghostkeeper, 2020-12-23:
If OpenGL 4.0 is not available it should fall back to OpenGL 2.0, which should be supported through backwards compatibility by your chip. Indeed you won't have the simulation view available but at least it should run and be able to see layer view data, albeit more flat and less fully-featured.

The crash dialogue indicates that there is a segfault (`c0000005`) in `ig4icd64.dll` which is the Intel graphics drivers. There should NEVER be a segfault in your Intel drivers. There is very little we can do to help with this, as it's a fault on Intel's side.
Other applications may not hit the same code path. There may be workarounds possible in theory to not hit the code path that is causing the segfault. But trying to work around segfaults in drivers on the users' computer is a bit like trying to throw a bottle in the ocean in the correct spot such that it lands in Hawaii by ocean currents. It's extremely unpredictable and we're basically flying blind.

mrx23dot, 2020-12-23:
Thanks for the investigation, would OpenGL 2.0 generate worse gcode?

Looks like the Windows Intel driver is buggy, I will try to upgrade it.
https://dslrdashboard.info/phpBB3/viewtopic.php?t=1572

Until then I have to stick with Cura 4.7.

fieldOfView, 2020-12-23:
OpenGL does not in any way affect the gcode that is created.

dorianboulch, 2021-02-15:
> With multiple monitors, try temporarily disabling one. If Cura starts with only a single display, then go to the General preferences in Cura and deselect the option to "Restore window position on start". Then you can reenable the second display.
The workaround worked for me.

This issue is still present in 4.8.0.

time4this, 2021-02-22:
What finally worked for me on 4.8 was to click on the **Learn More** of the last intro slide (before the crash)  and change the setting to not share info with Cura. Changing the setting and accepting the change allowed the installation to completed. Finally. Changing to a single monitor/powering off/disconnecting, etc. 

Good luck!

Ghostkeeper, 2021-03-02:
It's weird that for some people changing that preference apparently helps get them through the start-up wizard, especially since the anonymous slice information is _only_ sent when saving the g-code or sending it to the printer. The preference is not used in the start-up wizard at all. You can search the source code here for usages of the `info/send_slice_info` preference to find where it's used.

XanderStrike, 2021-03-29:
I was having this same issue _intermittently_ on macOS on my 2017 15-inch Macbook Pro, also seems related to Graphics Drivers. None of the above workarounds fixed it for me but I seem to have figured something out. For context, my MBP has Intel HD 630 graphics and a Radeon Pro 560 4 GB.

[image]

Given that it worked intermittently I took a close reading of the logs and found out that when it didn't work it used my Intel graphics:

```
2021-03-29 15:44:01,292 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2021-03-29 15:44:01,296 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1 INTEL-16.1.11
2021-03-29 15:44:01,300 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   Intel Inc.
2021-03-29 15:44:01,304 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: Intel(R) HD Graphics 630
2021-03-29 15:44:01,309 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.10.0
```

and when it did work it used my Radeon card:

```
2021-03-29 15:44:22,009 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2021-03-29 15:44:22,012 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1 ATI-4.2.15
2021-03-29 15:44:22,014 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   ATI Technologies Inc.
2021-03-29 15:44:22,017 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: AMD Radeon Pro 560 OpenGL Engine
2021-03-29 15:44:22,021 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.10.0
```

What's interesting is now after this debugging session constantly starting/stopping Cura with the different GPUs I am no longer able to repro the issue at all, it starts fine on both. I didn't go messing around in any settings so I'm not sure why that would be.

One can force their mac to always use discrete graphics (if available) by **unchecking Automatic Graphics Switching** in the power settings. For more granular control there's also [gfxCardStatus](https://gfx.io/) which lets you force integrated or discrete, and shows why it picked one or the other.

I'm happy to provide full logs or anything else needed for more clues.

Hope this helps!

jkovacs-dev, 2021-04-25:
> With multiple monitors, try temporarily disabling one. If Cura starts with only a single display, then go to the General preferences in Cura and deselect the option to "Restore window position on start". Then you can reenable the second display.

This worked for me in 4.9.0. Thanks @fieldOfView!

bn326160, 2021-05-12:
> One can force their mac to always use discrete graphics (if available) by **unchecking Automatic Graphics Switching** in the power settings. For more granular control there's also [gfxCardStatus](https://gfx.io/) which lets you force integrated or discrete, and shows why it picked one or the other.

Thank you, I got this issue after upgrading from  macOS 11.1 to 11.3.1!
I have a 2018 MacBook Pro 15" with a Intel UHD630 & Radeon Pro 560X

grimmaldus, 2021-06-28:
> @Devtech223 I hadn't seen that one, thanks for pointing it out! Are your 'reproduction steps' the same as @MooM00 s?
>
> I'm going to close the newer issue #8904 as a duplicate (keeping this one open [...]

I can confirm it was due to two monitors.

nallath, 2021-06-28:
We've made some changes to 4.10 that might affect this issue. A number of people have reported that issues with multiple monitors have been resolved in 4.10

Mfespinosa92, 2021-08-28:
Had the same issue. I have an i7 late 2014 Mac mini with Dual Thunderbolt Displays along with an EGPU attached. Somewhere in some update using the EGPU to render vs the internal gpu causes the white screen. Disabling "prefer EGPU" restored Cura back to normal. Cura upgrades did not resolve the issue. I used to render using my egpu previously with no issues.

Inzaniity, 2022-05-13:
Hey, I encounter Cura crashing on me too, tried to disable screens and launch it but didn't help. Deleted the cura.cfg and that didn't help either. 

Here's the log: 

```2022-05-13 23:34:53,229 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [87]: Adding QT6 plugin path: C:\Program Files\Ultimaker Cura 5.0.0\PyQt6\plugins
2022-05-13 23:34:53,229 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [97]: Adding QT5 plugin path: C:\Program Files\Ultimaker Cura 5.0.0\PyQt6\plugins
2022-05-13 23:34:53,229 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2022-05-13 23:34:53,229 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 5.0.0-beta+1
2022-05-13 23:34:53,230 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 8.0.0
2022-05-13 23:34:53,230 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
2022-05-13 23:34:53,230 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib\uranium must be a folder.
2022-05-13 23:34:53,230 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib64\uranium must be a folder.
2022-05-13 23:34:53,230 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib32\uranium must be a folder.
2022-05-13 23:34:53,230 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Ultimaker Cura 5.0.0\plugins must be a folder.
2022-05-13 23:34:53,231 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Resources\uranium\plugins must be a folder.
2022-05-13 23:34:53,231 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Resources\cura\plugins must be a folder.
2022-05-13 23:34:53,231 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
2022-05-13 23:34:53,231 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is C:\Users\Jan\AppData\Roaming\cura\5.0
2022-05-13 23:34:53,231 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is C:\Users\Jan\AppData\Roaming\cura\5.0
2022-05-13 23:34:53,231 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is C:\Users\Jan\AppData\Local\cura\5.0\cache
2022-05-13 23:34:53,232 - INFO - [MainThread] UM.Qt.QtApplication.initialize [149]: Preferences file not found, ignore and use default language 'en_US'
2022-05-13 23:34:53,232 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.0.0\share\uranium\resources\bundled_packages\uranium.json
2022-05-13 23:34:53,232 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.0.0\UM\..\share\uranium\resources\bundled_packages\uranium.json
2022-05-13 23:34:53,232 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.0.0\cura\..\share\cura\resources\bundled_packages\cura.json
2022-05-13 23:34:53,232 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\bundled_packages\cura.json
2022-05-13 23:34:53,232 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2022-05-13 23:34:53,232 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2022-05-13 23:34:53,233 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [180]: Detected most suitable OpenGL context version: 4.1 Core profile
2022-05-13 23:34:53,233 - INFO - [MainThread] UM.Qt.QtApplication.initialize [187]: Initializing job queue ...
2022-05-13 23:34:53,233 - INFO - [MainThread] UM.Qt.QtApplication.initialize [191]: Initializing version upgrade manager ...
2022-05-13 23:34:53,233 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 5.0.0\share\uranium\resources\bundled_packages\uranium.json
2022-05-13 23:34:53,233 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 5.0.0\UM\..\share\uranium\resources\bundled_packages\uranium.json
2022-05-13 23:34:53,233 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 5.0.0\cura\..\share\cura\resources\bundled_packages\cura.json
2022-05-13 23:34:53,233 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\bundled_packages\cura.json
2022-05-13 23:34:53,233 - INFO - [MainThread] UM.PackageManager._loadManagementData [203]: User package management file C:\Users\Jan\AppData\Roaming\cura\5.0\packages.json doesn't exist, do nothing
2022-05-13 23:34:53,234 - INFO - [MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - 
[... cut ...]
2022-05-13 23:34:58,737 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\xray_composite.shader]...
2022-05-13 23:34:58,739 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\uranium\resources\shaders\platform.shader]...
2022-05-13 23:34:58,743 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\xray.shader]...
```

Happens with 5.0 and 4.13

liera21, 2022-10-23:
> **FIXED IT BY STOPPING ASUS GPU TWEAK SOFTWARE THAT CAME WITH MY TUF 3060TI!** Even though Cura ran perfectly on the same machine with the same software a few days ago...

Hi, I just created my github account to thank and confirm this solution, I've been having the same issue (Cura closing after loading every time) with a previous and the latest Cura versions, closed GPU Tweak II and Cura is running fine again. 

Then it occurred to me that the OSD overlay might be trying to interact with the Cura window, and indeed, I can confirm (at least in my case) that the problem is the GPU Tweak OSD overlay, I have Cura running with GPU Tweak but the OSD overlay off. I hope this helps.

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

dorianboulch, 2024-11-20:
IMO, an update solves it, so it could be closed.

GregValiant, 2024-11-20:
Thanks.  I'm cleaning house here.

closed 2024-11-20 by GregValiant