Ultimaker/Cura #8894: propagate retraction to idle extruders In multiple filament single nozzle hotend

edolis, 2020-12-07:
On multiple filament bowden  - single nozzle printers, when printing with one filament out of three for instance, retraction is not sufficient to create enough suction to prevent oozing and stringing
I believe that is caused by retraction volume being compensated by the fused filament from the inactive filaments running into the chamber

**Describe the solution you'd like**
being able to propagate retraction to inactive extruders by a percentage of the retraction from the filament with highest mix percentage (see example below)

**Describe alternatives you've considered**
Usually I increase at maximum the retraction acceleration on the extruder, but this is not enough and spreading retraction would help to get better results with lower acceleration.

I think this would be extremely beneficial to all user having multiple filament - single nozzle hotends

**Additional context**
Example of how this could work
[image]

Ghostkeeper, 2020-12-09:
Shouldn't the `M567` or `M163` command already do this?

I don't see how Cura could implement this, because

* G-codes are only executed one at a time. We can't move two feeders at the same time, since the printer will wait for the first retraction command to be completed before it starts with the next retraction command.
* `G0` and `G1` don't have any parameter that indicates which feeder gets moved. We could switch extruders before issuing the command, but on some printers that triggers a whole extruder switch procedure that you definitely don't want to happen during a retraction.

This basically has to be implemented by firmware, since g-code is not powerful enough to implement your feature request.

closed 2020-12-09 by Ghostkeeper

edolis, 2020-12-09:
Thanks for the reply, understood!