Ultimaker/Cura #8898: Arrange Models does not compensate for rotation

KenLucke, 2020-12-07:
**Application version**
4.8 (and before)

**Platform**
Mac OSX 10.15.7

**Printer**
N/A

**Reproduction steps**

1. Load a model that does not fit into build space in one orientation, but does in another.
2. Use Arrange Models to try to automatically rotate to fit

**Screenshot(s)**
Notice the initial position is non-tenable (although I could change some build plate parameters to compensate).  Manual rotation of 22° makes it allowable.

[image]

**Actual results**
Nothing - no change

**Expected results**
Arrange Models should figure out a way to fit  the model if at all possible. 

**Project file**
https://www.dropbox.com/s/xfwdysorkk6inrr/Meteor%20Light%20Holder.3mf

**Log file**

**Additional information**

rburema, 2020-12-08:
'Arrange' doesn't perform rotation, and is mostly intended to be used if there are multiple objects on the build-plate. (The 'bin-packing' problem is already NP-complete even without rotation).

I consider this a feature request then: Make the arrange algorithm be able to rotate models. This will be very hard in the general case (with more than one object), so I'll just discuss a 'fit single object to build-plate' that takes rotation into account with my colleagues.

KenLucke, 2020-12-08:
> 'Arrange' doesn't perform rotation (EDIT: _now_, except in 90 degrees), and is mostly intended to be used if there are multiple objects on the build-plate. (The 'bin-packing' problem is already NP-complete even without rotation).
> 
> I consider this a feature request then: Make the arrange algorithm be able to rotate models. This will be very hard in the general case (with more than one object), so I'll just discuss a 'fit single object to build-plate' that takes rotation into account with my colleagues.

OK, works for me :) .

KenLucke, 2020-12-08:
Also, related, see #8899

JRitmeester, 2021-11-24:
I just came across this, as I have models that only fit at an odd angle between 45 and 90 degrees on my build plate. Whatever came out of this discussion?

rburema, 2021-11-30:
From memory, as I can't find the related ticket: It was discussed, but deferred. as adding rotation in anything other than 90 degree increments would get prohibitively expensive (read: will take a long time to run) for the situations where it's needed (many small models on a large buildplate).

However. there _is_ a ticket on the internal backlog (devs: CURA-7951) to 'lock' the orientation of the models when arranging.