Ultimaker/Cura #8903: Any new G-code files wont print on 3D printer

JpTech49, 2020-12-08:
To be quick, I have a problem where my printer won't read any new g-code files

I currently am running the 4.8.0 version of Cura on my Ender 5pro 3d printer and ran into this problem last night. finding that my printer just stops at around in the middle of layer 201. Thinking there was a blackout last night, I ran the same g-code and got the same answer.

When I tried to use a different g-code, it only shows a "9s" on my screen of the printer. With other new g-codes, it only showed variants like "4s" or "7s".  older G-codes before this update works perfectly fine, but I cant seem to get anything new to even start printing on my printer.

I am currently testing to see if all of my older g-codes work and if there is anything on my end of setting up G-codes
but I have already done the following
- re formatted my USB
- remade the probomatic files
- backup my current files and full on reset the USB in question
- tested with new g-codes that haven't previously been used on my printer to confirm that it is affecting any new file

rburema, 2020-12-08:
Is there a post-processing script on? (There is a little icon with a hammer and wrench next to the slice button area in that case.)

Ghostkeeper, 2020-12-09:
It would be useful to see the g-code that is not accepted by the printer, and a project file that generates this faulty g-code.

closed 2020-12-23 by no-response[bot]