Ultimaker/Cura #8907: [4.8] Cura often switches my profile settings between restarts

ubershy, 2020-12-08:
**Application version**
4.8

**Platform**
Windows 10 x64, GTX 1070, Ryzen 5 2700x.

**Printer**
Anet A6 (custom definition file, was installed before the official definition became available).
On the video it is called "AnetA6_0.8".

**Reproduction steps**
1. Install and use my configuration folder instead of yours, which I attached to this bug report. My configuration folder is needed to reproduce the bug.
2. Start Cura.
3. Take a look at the current value of "Wall line count".
4. Close Cura.
5. Start Cura.
6. Take a look at the current value of "Wall line count" setting - it is now different. With some probability it will not different, if so go to step 2 and repeat. 

**Screenshot(s)**
Example.
Before restart:
[image]
After restart:
[image]

**Actual results**
"Wall line count" setting is often changed between restarts with some probability.

**Expected results**
"Wall line count" setting should stay the same between restarts.

**Project file**
Not needed.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5662336/cura.log)

**Additional information**
Video demonstrating the problem: https://youtu.be/VQTtASpuVQY

My configuration folder which is needed for reproduction:
[cura.zip](https://github.com/Ultimaker/Cura/files/5662345/cura.zip)

I don't remember when exactly the problem started to happen, maybe since version ~4.3-4.7. I haven't printed much since, waiting for the problem to be fixed with the each next release. Note that I have been using Cura for a long time across many versions.

I also tried turning off all non-bundled plugins (including plugins related to profiles), but the problem is still happening.

I suppose my configuration might be corrupted or containing _something_ that makes the current cura behave incorrectly.  
**If so, what is wrong with my configuration and how can I fix it?** What should I do in the future to prevent this from happening? I don't want to recreate all my profiles from scratch only to corrupt somehow my configuration again.

Please help.

nallath, 2020-12-09:
Well, it's probably not what you want any developer to say to you, but "Woah, that is an interesting one".

But since you provided us with quite a bit of info, I think I have a pretty good idea how this happend (although I'm not a 100% sure!).

What is happening is that there are 2 extruder files "fdmextruder+%233.extruder.cfg" and "fdmextruder+%235.extruder.cfg" that point to the same machine "AnetA6_0.8". Cura expects there to only be one. Pretty much all the code just takes the first one that it encounters (which is fine, since it should only ever get one!). Since the order in which these are returned is somewhat random, you get different results every time (One of the extruders sets the wall line count to 1, the other to 21. So that's why the value changes!)

As to how this happened; I don't fully know. I think it might have been caused by an upgrader script, but since you are the first to report this, it's a bit hard to find out where exactly.

As for fixing the issue; Removing either of the files (in your config in the extruder folders) should do the trick!

ubershy, 2020-12-11:
@nallath 
Thank you very much for such detailed answer. The trick indeed worked!

I think I'm okay not knowing how this happened, at least now I will know what to look for next time I get a similar problem.

> Well, it's probably not what you want any developer to say to you, but "Woah, that is an interesting one".

Yeah, sorry... I guess your main concern is that I have too many quality profiles.  
Since the Materials in Cura could not contain the settings I wanted them to contain, I decided to store all material-related settings (except "Diameter") in Quality Profiles. So I made a separate Quality Profile for each material type.
And for a long time I printed with 0.4 nozzle. When I started using also 0.8mm nozzle, I noticed that I need to print with slightly different settings. And so I created an additional set of Quality Profiles for 0.8mm nozzle. This increased the number of quality profiles 2x further. In mathematics it is called "combinatorial explosion".

I think I can remove most of the Quality Profiles now, because there is "Material settings" plugin now available.

nallath, 2020-12-11:
Oh no, i didn't make a reference to your quality profiles. It was towards the "huh, how did it even manage to get into this state".

I'll close this issue. If you manage to reproduce it, feel free to re-report it!

closed 2020-12-11 by nallath