Ultimaker/Cura #8908: [4.8.0] prime tower Y position is negative with respect to reality

bellzw, 2020-12-08:
**Application version**
Cura 4.8.0

**Platform**
Windows 10 64-bit running on Lenovo Yoga 14

**Printer**
BIBO Dual

**Reproduction steps**
1. Open model stl
2. Lay model flat on build surface, if necessary
3. Move model to (0, -25, 0)
4. Copy model and place at (0, 25, 0)
5. Select 1st copy and assign extruder 1 to it
6. Select 2nd copy and assign extruder 2 to it
7. Place prime tower at 60, -20

**Screenshot(s)**
[image]

**Actual results**
Prime tower is actually at Y=+20.  Cura appears to use the negative of the Y Position as the location of the most positive Y position of the tower.

**Expected results**
Cura should use Prime Tower Y Position to define the most positive Y position of the tower. 

**Project file**
[Prime Tower position.zip](https://github.com/Ultimaker/Cura/files/5662448/Prime.Tower.position.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
It would actually be better, I think, if Prime Tower X and Y position defined the center of the tower rather than the extrema.

Ghostkeeper, 2020-12-17:
The issue here is a bit bigger. Cura has a coordinate system which has the origin always in the centre and positive Y towards the front of this printer. But the g-code also has a coordinate system which is printer-specific. In your case you happen to have the origin also in the centre (not the case for most printers) but then still with a flipped Y coordinate.
If the origin is in the bottom left, the issue is a bit less confusing because the coordinates are more wildly different.

I think it's important that Cura has a consistent coordinate system for all printers. For one, it's handy to be able to centre an object by setting its position to 0,0 rather than 115.5,120 or whatever the half of the build plate size is. It's also easier to switch between printers then without having to get used to a different coordinate system.

The problem is that this Cura-specific coordinate system is not used for settings which indicate coordinates, like the Layer Start X/Y, the seam position, prime blob position or the prime tower position. Those use your printer's coordinate system. This was not an explicit choice I don't think. It's just that CuraEngine doesn't know about the coordinate system of Cura's front-end and the front-end doesn't translate it for CuraEngine either.

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

bellzw, 2024-11-21:
Yes, I think this one can also be closed.  I haven’t printed something using both extruders with 5.9.0, yet, because I need to replace my E2 nozzle.  So, to answer your question quickly, I generated a gcode file with 2 objects and assigned them to E1 and E2, enabled a prime tower, and  sliced.  Looking at the gcode file, I see that the coordinates in the file for the prime tower use the same coordinate system as for the objects to be printed, and the tower would print as it shows in Cura.

 

There does seem to be a few mm discrepancy between the coordinates in the gcode compared to what I see in Cura, but that’s a separate issue and I’ll start a new thread.

bellzw, 2024-11-21:
Never mind about the discrepancy between the prime tower coordinates in the gcode file and what I see in Cura 5.9.0.  I missed a line of code in the file that explains my confusion.  It’s all good.

GregValiant, 2024-11-21:
What I consider an oddity (old draftsman here) the Prime Tower location is not to the centerline but rather to the periphery (the right edge and the edge closest to the front of the build plate).  That always seemed odd to me.  

Here I have the model at the build plate mid-point and the prime tower location is also at the build plate mid-point but the prime tower is offset back and left by it's radius.
[image]

If I double the size of the prime tower the centerline moves back further and left further.  In this example, the model is now printable.  Without changing the Prime Tower location...the Prime Tower location changed (in my opinion).
[image]

There was a post here somewhere that made a quick change to a couple of files and resulted in the "Model Origin" flipping the Y so that it agreed with the build plate +Y.  I don't know if there might be "unexpected consequences" but I haven't been bitten yet.
[image]

GregValiant, 2024-11-21:
I wish my confusion could be explained so easily.
I'll go ahead and close this then.

bellzw, 2024-11-21:
> What I consider an oddity (old draftsman here) the Prime Tower location is not to the centerline but rather to the periphery (the right edge and the edge closest to the front of the build plate). That always seemed odd to me.

You might notice my comment in the Additional Information at the top of the issue.  This has been the case for a long time and not at all intuitive.  It would be nice it the specification was the center of the tower.

BTW, I still have my French curves, triangles, and protractors from my mechanical drawing days before computers became ubiquitous. :-)

GregValiant, 2024-11-21:
_"I still have my French curves, triangles, and protractors..."_
Me too, and I have a bandsaw.  That gives me all the tools for laying out and cutting down motorcycle windscreens.

About 40 years ago I broke my 2meter scale.  I kept the big piece of it but it's funny how when I see it, it still pisses me off.