Ultimaker/Cura #8910: [4.8.0]Prime tower/prime tower brim shadows are generated when extruder 2 is the only extruder used and prime tower is enabled

bellzw, 2020-12-08:
**Application version**
Cura 4.8.0

**Platform**
Windows 10 64-bit running on a Lenovo Yoga 14 laptop

**Printer**
BIBO Dual

**Reproduction steps**
1. Open attached 3mf
2. Enable prime tower or both prime tower and prime tower brim
3. Select an object and assign extruder 2 to it
4. Select an object and assign extruder 2 to it

**Screenshot(s)**
[image]

**Actual results**
Prime tower and/or brim shadows appear and can be recognized as an interference, even though they not apparently part of the build (move the prime tower to see this).  This does not happen if both objects are assigned to extruder 1; neither shadow is present in this case.

**Expected results**
When all objects are assigned to the same extruder, the prime tower and/or brim shadows should not be in the scene if the prime tower/brim are not actually part of the build, regardless of which extruder is chosen.

**Project file**

[Prime Tower brim shadow.zip](https://github.com/Ultimaker/Cura/files/5662508/Prime.Tower.brim.shadow.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**

rburema, 2020-12-15:
Thanks for the report. We have an internal ticket (CURA-7908) that we have slated for the next two sprints somewhere that we think _might_ be the cause of this. I've made a note in that ticket, so if it doesn't fix it, we can redetermine the priority,

Ghostkeeper, 2020-12-17:
It's correct that a prime tower shadow is shown, because:
* The models are printed with extruder 2.
* The build plate adhesion is printed with extruder 1. Indeed, the build plate adhesion type is set to None, but there is still a Prime Tower Brim so that is build plate adhesion and therefore it thinks extruder 1 is used.

Since both extruders are being used and the Prime Tower is enabled, there is a shadow that reserves space for the prime tower.

There is an inherent circular dependency in this. Which extruders are used depends on the build plate adhesion. Whether there is build plate adhesion depends on the prime tower. Whether there is a prime tower depends on which extruders are used. I'd say that the real bug here is that the front-end and CuraEngine resolve this circular dependency differently. In this case CuraEngine resolves it more efficiently by not printing the prime tower.

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

bellzw, 2024-11-21:
Yes, it can be closed.  With a newly installed 5.9.0, putting the infill, walls, support, adhesion, top/bottom all  on 2, the prime tower shadow disappears, and extruder 1 is not used in the gcode file.

closed 2024-11-21 by GregValiant