Ultimaker/Cura #8911: Z Seam Alignment Random always cura always starts on a corner

nathanshearer, 2020-12-09:
**Application version**
Cura 4.8.0

**Platform**
Gentoo Linux, using the official AppImage

**Printer**
Creality Ender 3

**Reproduction steps**
1. Enable "Z Seam Alignment: Random"
2. Slice a cylinar
3. Notice that the start of each wall is always a random location anywhere around the outside wall
4. Slice a cube or rectangle or anything with sharp corners
5. Notice that the start of each wall is always a randomly chosen corner, not a true random start anywhere along the outside wall. It is _always_ a corner. 

**Expected results**
I expect that the wall should very rarely actually start on a corner for most objects that have only a few corners. Instead the start and stop of every single layer is always a corner. And since under/over extrusion is most evident at the start/end of a line, this results in corners being very inaccurate.

nathanshearer, 2020-12-09:
Here is a calibration print where there are 4 corners on the outside wall. Notice that the "random" starts are always on the corners, resulting in over/under extrusion being amplified at those locations.

[image]

Here is a mostly round STL where the random starts are working great.

[image]

nathanshearer, 2020-12-09:
Here is a 20mm cube with 90-segment 5mm beveled edges, in an attempt to find out when Cura starts preferring to start at corners. It seems that the "random" start has a preference for sharper regions of a line instead of what I would expect: a true random start anywhere along the wall.

[image]

Ghostkeeper, 2020-12-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/8010.

closed 2020-12-18 by Ghostkeeper