Ultimaker/Cura #8912: Request: Move Initial Layer Line Width to Platform Adhesion

whosawhatsis, 2020-12-09:
**Is your feature request related to a problem? Please describe.**
The purpose of the Initial Layer Line Width setting is intended to help with platform adhesion. The original idea was to print a raft-like layer on the bottom of a print that wouldn't need to be peeled away after printing. It works well for this with values up to 200%, but this is not compatible with some other platform adhesion options, particularly rafts. When you print with a raft, you have to remember to also set it back to 100%, or the raft won't come off.

**Describe the solution you'd like**
The setting really should be disabled (or combined with the Raft Base Line Width setting) when rafts are enabled. The setting, along with Initial Layer Height, would make a lot more sense in the Platform adhesion section, as both of these options are really about platform adhesion, rather than quality.

**Describe alternatives you've considered**
When using this setting for platform adhesion purposes, if you then have a print that instead needs a raft, you need to either:
- Remember to change both settings, or
- Maintain an alternate, parallel version of each print profile with rafts enabled and an Initial Layer Line Width of 100%.

**Affected users and/or printers**
Anyone using Initial Layer Line Width (for its intended purpose), particularly if they occasionally want to print with a raft instead (such as for prints with sections of the base that are too narrow for the wider lines).

Ghostkeeper, 2020-12-18:
The Initial Layer Line Width is put in the quality category to be next to the other line width settings. After all, if you change the Top/Bottom Line Width you might want to also adjust the Initial Layer Line Width or else you get lines that are too wide to print. This is a similar problem as what you're describing. If you change one setting, you need to remember that you might want to change the other too.

The setting categories currently slot settings based on what structure the settings are part of more than what print problem they intend to solve. If it were based on the problems you want to solve, the Build Plate Temperature and Initial Layer Bottom Pattern are also definitely in the build plate adhesion category. Your confusion is, I think, that the build plate adhesion structure (brim/raft/skirt) is named after the problem it wants to solve, while other categories (shell, infill, support, etc) are not. And there are some categories which are not named after a physical part of the print at all (Quality and Experimental).

In general, I think it would be better to move the Line Width settings into their appropriate categories (Infill Line Width into the infill category, etc.), which would place Initial Layer Line Width into the Shell category. But there are valid arguments for either side. It's a bit bikeshed-y.

whosawhatsis, 2021-01-12:
To your point about changing it when you change other line width settings, that wouldn't be an issue if it was an absolute value, rather than a percentage.

Unlike other width/thickness settings that are primarily used to affect the tradeoffs between print quality and speed (while having secondary effects on things like print strength), the ONLY reason you need settings specifically for the line width/layer thickness of the first layer is for their effects on platform adhesion. Initial layer speed is really in the same bucket. It would be really nice if there was some way to create ad-hoc groups of settings like these, taken from different categories, and show them next to each other, perhaps in a "custom" settings group. Being able to apply multiple profiles to a print at once (e.g. one for quality, and a second profile containing settings for platform adhesion) would be a nice option, though I imagine it would be tricky to implement in a user-friendly way (MatterControl has tried to do this, with mixed results).

**How about this as a compromise that should work better for everyone:**

Under buildplate adhesion, there are a number of settings for the thickness, line width, and speed of various parts of the raft. If this section included settings for the initial layer printed ON TOP of the raft, which would be used instead of the regular initial layer settings when raft is enabled (after all, that isn't really the initial layer of the print anymore when raft is on). If that setting inherited its defaults from the existing initial layer settings, existing profiles should be unaffected by the change, and customizing these new settings would make it necessary to mess with other settings when enabling raft.

Ghostkeeper, 2021-01-25:
There are indeed many cases where a setting needs to be changed when another setting is changed, in particular for the settings which choose from different types of strategies, like Adhesion Type. I guess the most common one is Enable Support, which require you to choose different settings for the skin because they are then resting on support rather than floating in the air.

Profiles in Cura can choose different settings automatically, based on the values of other settings, like:

    initial_layer_line_width_factor = =120 if adhesion_type != "raft" else 90

This would choose 90% if the initial layer of the print is on a raft, or 120% if it's on the build plate.

Only the user can't enter these formulas in the GUI (yet).
I don't really think it's a good option to create settings for every case where a user might want to choose a different value automatically based on other settings. There would be many. And so far, the Initial Layer Line Width is not adjusted automatically in any profile that Cura ships with, so Initial Layer Line Width wouldn't be the first candidate.
It wouldn't be a lot of work to implement such settings, on the other hand. But generally we try to not worsen the problem too much of the users not being able to find the correct settings to modify.

whosawhatsis, 2021-02-19:
Is there a reference for that syntax anywhere? I've been trying to set that up in my custom machine definition (the JSON file). I already have a few conditional settings in there (based on checkbox settings like magic_spiralize), and ones that are functions of another setting, but the correct syntax for a conditional based on a setting like adhesion_type in the JSON is eluding me.

Ghostkeeper, 2021-03-01:
It's a Python expression (documented [here](https://github.com/Ultimaker/Cura/blob/master/docs/profiles/getting_a_setting_value.md#evaluating-a-value-property)), so the reference would be [here](https://docs.python.org/3/reference/simple_stmts.html#expression-statements) but nobody actually reads that. Unfortunately, the ternary operator in Python is as confusing as a dead Vulcan on Mars.

My best summary for the ternary operator is:

    <main value> if <condition for main value> else <fallback>

So in my example above, it the main value is 120% line width factor. The condition for the main value is that it's not using a raft. If that condition fails (in the case of the raft), it uses the fallback of 90% initial layer line width.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant