Ultimaker/Cura #8913: Monitor tab doesn't show current progress nor items in queue

chriskewl, 2020-12-09:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
Ultimaker 3

**Reproduction steps**
1. Print something
2. Once done, print something else
3. Notice the item still shows the original item printed with no update on progress.
4. You can still see correct status if you "Manage in browser".

**Screenshot(s)**
Cura Monitor tab: https://cdn.discordapp.com/attachments/316634826407149569/786163615245926420/unknown.png
Browser Connect: https://cdn.discordapp.com/attachments/316634826407149569/786163636980678656/unknown.png

**Actual results**
Notice the item still shows the original item printed with no update on progress. It also doesn't show anything in the queue.

**Expected results**
Should show the progress of the currently printed object with any items in the queue.

Ghostkeeper, 2020-12-18:
It would be good to have separate issues for different bugs.

I'm not able to reproduce this issue. I can see different items in the queue fine:

[image]

Maybe the firmware version on the printer is outdated? What version are you using?
It kind of sounds like Cura has just lost connection to the printer.

chriskewl, 2020-12-23:
I uninstalled Cura and reinstalled it. Still no change. Also Cura is sending items to the printer just fine. Firmware is at latest version.

Ghostkeeper, 2020-12-28:
I still haven't seen this issue, though I use my printer pretty much every workday.

Perhaps other people have seen it too. Then we can find the commonalities between your set-ups. Maybe it's something network-related? Or maybe the API on the printer is not functioning reliably?

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

chriskewl, 2024-11-20:
> Is this still an issue in current Cura versions (5.8.0 and up)? Can this be closed?

You can close it for now, we'll see if this comes back.

closed 2024-11-20 by GregValiant