Ultimaker/Cura #8915: Unable to open any file, Crashing, Cura 4.8

bscott1969, 2020-12-09:
**Application version**
4.8

**Platform**
OSX with AMD RX Vega 56 

**Printer**
AnyCubic Mega

**Reproduction steps**
1. opened program
2. opened previous file, tried various files and checked printer profile that there were no = in the config

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Crash

**Expected results**
to be able to print the file

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)
** Trackback **
Traceback (most recent call last):
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1063, in event
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 464, in event
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/UM/Event.py", line 218, in call
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332, in __performEmit
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1825, in _readMeshFinished
IndexError: list index out of range

**Log file**
Thread 0x0000700013e2c000 (most recent call first):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700013929000 (most recent call first):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700013426000 (most recent call first):
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/brianscott/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 2184 in wait
  File "/Users/brianscott/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 1496 in run
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700012f23000 (most recent call first):
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/brianscott/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 2184 in wait
  File "/Users/brianscott/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 1315 in run
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700012a20000 (most recent call first):
  File "/Users/brianscott/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 1221 in run
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700012394000 (most recent call first):
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 1434 in run
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
[... 98 lines cut ...]
Current thread 0x0000000112177e00 (most recent call first):
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/Users/ultimaker/build/4.8/build/inst/bin/cura_app.py", line 167 in exceptHook
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/Users/ultimaker/build/4.8/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 847 in run
  File "/Users/ultimaker/build/4.8/build/inst/bin/cura_app.py", line 235 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/Users/ultimaker/build/env/4.8/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-12-09:
Could you also share your configuration folder? 

The issue with the "=" has been fixed, but there has been a moment where it could corrupt your setup in such a way that it couldn't really recover from it.

bscott1969, 2020-12-09:
bscott@btinternet.com sent you some files

1 item, 11.8 MB in total ・ Expires on 16 December, 2020

Download link:
https://wetransfer.com/downloads/4a5f87cbb6f035964899b42561982ca220201209133716/5f1704ca1a7e3f6a1cc564abcd43377020201209133731/78e235

Message:
The contents of my Config folder

1 item

4.8.zip - 11.8 MB

[Get more out of WeTransfer, get Pro](https://wetransfer.com/pro?trk=WT201704_email&utm_campaign=WT_email_tracking&utm_content=general&utm_medium=plus_footer_ad_link&utm_source=notify_recipient_email)

About WeTransfer: https://wetransfer.com/about
Help: https://wetransfer.zendesk.com/hc/en-us
Legal: https://wetransfer.com/legal/terms

To make sure our emails arrive, please add noreply@wetransfer.com to [your contacts](https://wetransfer.zendesk.com/hc/en-us/articles/204909429).

nallath, 2020-12-09:
I've downloaded your config folder and installed it. It's indeed what I thought it would be; The anycubic configuration is corrupted. 

If you remove the anycubic and re-add it again, the problem should be fixed!

closed 2020-12-09 by nallath

tstan93, 2020-12-17:
I just downloaded cura, and seem to be having a similar problem. Upon opening a file to start slicing the program crashes. I have tried exporting, file open, and the icon on the screen all do the same thing.

Any ideas?

Ghostkeeper, 2020-12-18:
Try deleting your printer from the list of printers and adding it back. The problem occurred if you were using version 4.7.0 and had an `=` symbol in your start g-code.

tstan93, 2020-12-18:
> Try deleting your printer from the list of printers and adding it back. The problem occurred if you were using version 4.7.0 and had an `=` symbol in your start g-code.

I figured it out, i ended up needing to go through and disabling the dell recovery and backup extension like I had seen in a previous thread. Thanks for the help Ghostkeeper.

dearmonds, 2020-12-30:
I have the same issue, but cannot resolve.  Have tried reinstalling 4.8.    Any attempt to Open a file causes program to crash/close.  I have a Dell XPS 8700 with i7 running Windows 10 Pro.   Creality Slicer on Ender3 flashstick works fine.    Not sure if I even need this upgrade but decided to try it, and am running into bugs right off the bat so I can see why Creality isn't anxious to put the latest Cura software on the flashstick!  My faith in Cura is not good right now.

Ghostkeeper, 2021-01-08:
@dearmonds Do you have a log file for us, taken just after it crashes?