Ultimaker/Cura #8919: [4.8] Manual support plugin causes cura model absolute positioning errors

Liger0, 2020-12-09:
**Application version**
4.8

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Use tree support
2. Place manual supports with plugin
3. A manual support didn't touch the object (a plugin bug?)
4. Change the manual support dimension.

**Screenshot(s)**
[image]

**Actual results**
The eventual bug with the plugin caused an eventual bug with the cura positioning coordinates.
When I enlarged the support on the Z, the model to print went high from the plane, but the Z coordinate on cura didn't update accordingly.

**Expected results**
Z coordinates should account edge case suppport behavious.

**Project file**
[CCR10S_manual support with tree support changes positioning.zip](https://github.com/Ultimaker/Cura/files/5669152/CCR10S_manual.support.with.tree.support.changes.positioning.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-12-16:
If the error/bug/issue is within the plugin, you might want to notify the author of the plugin instead.

Info about that plugin, including the authors' email, can be found here: https://marketplace.ultimaker.com/app/cura/plugins/lokster/CustomSupports

Liger0, 2020-12-16:
> 
> 
> If the error/bug/issue is within the plugin, you might want to notify the author of the plugin instead.
> 
> Info about that plugin, including the authors' email, can be found here: https://marketplace.ultimaker.com/app/cura/plugins/lokster/CustomSupports

As I said, the support placement may be a plugin bug, but I doubt the Z positioning and reading is.

Ghostkeeper, 2020-12-18:
Probably the same problem as https://github.com/Ultimaker/Cura/issues/8878.

closed 2024-12-08 by github-actions[bot]