Ultimaker/Cura #8920: [4.8.0] Object created when importing and image is not placed at origin

bellzw, 2020-12-09:
**Application version**
Cura 4.8.0

**Platform**
Windows 10 64-bit running on Lenovo Yoga 14

**Printer**
BIBO Dual

**Reproduction steps**
1. Open png image
2. set base = 0.8, height = 4.4, Linear mode, darker is higher, width = 101.6, depth = 101.6
3. slice
4. click on the object

**Screenshot(s)**
[image] - small offsets shown in upper left.
 
[image] - original image to convert

**Actual results**
Object is at small x, y offset, z= 0

**Expected results**
x = y = z = 0

**Project file**
[object not at origin.zip](https://github.com/Ultimaker/Cura/files/5669164/object.not.at.origin.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
Offsets are tiny and possibly caused by round-off error.  I doubt they make any difference to any printer.   However, if this is because of an uninitialized variable, then unexpected worse problems might occur down the line.

rburema, 2020-12-11:
I spent some time looking into this, and while it doesn't reproduce on my end (I just get 0,0,0 in those fields), it's not the result of uninitialised variable(s).

Since that seems to be the only real concern, I'll close this issue now. Feel free to reply if you think this isn't enough of a resolution.

closed 2020-12-11 by rburema

bellzw, 2020-12-11:
Odd that you can’t reproduce it.  It happens on my laptop every time, whether it’s an image or an stl.  The (x,y) values are always 0.0031 mm or -0.0031 mm, and they remain even after I rotate around the x or y axis to lay the part flat on the build plate.  Maybe it’s roundoff error?   I mentioned an uninitialized variable only because I’ve made such mistakes programming myself.

 

I find it hard to believe that it’s specific to the Lenovo Yoga 14, but I’m afraid that’s the only computer I have, so I can’t test this hypothesis easily.

 

But, the offset is very small and it’s revealed when I click on the part, so I can always force it to 0,0.  I don’t think I saw this in previous versions of Cura.  Is there a setting somewhere for the displayed precision in text boxes?  Maybe your installation isn’t showing a sufficient number of digits?

 

I have the following plugins installed:

 

I checked and Randomize Location is NOT checked in Mesh Tools.

 

Thanks for looking into this.

 

Zane

Ghostkeeper, 2020-12-18:
I've seen before that it doesn't always place the object exactly in the centre when loading a mesh (not specific to meshes created from images). It's sometimes off by a few micrometres. This is probably indeed a rounding error introduced by libnest2d which is used to position these objects. It got introduced when we transitioned to the new arrange algorithm in 4.8.

I don't think it's really a problem, practically. It's just that the field has this tiny fractional number in it. Your printer is unable to position an object within 3 micrometres accuracy anyway.

I can't seem to reproduce the problem now either, not with my own models nor with your image. Maybe something was changed here since the release of 4.8.