Ultimaker/Cura #8921: [4.8] Support blocker can only enlarge the body

Liger0, 2020-12-09:
**Application version**
4.8

**Platform**
Windows 10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Place a support blocker
2. Try to resize it manually

**Screenshot(s)**
https://youtu.be/0agYOKUuMDE

**Actual results**
You can't make it smaller, and making it larger is a challenge as it isn't a proportional enlargement compared to mouse drag

**Expected results**
Fluid resizing

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-12-16:
I can't reproduce this (also on Win10 x64).

Idea: Do you have 'screen scaling' on? It _should_ still work _with_ screen scaling on, but it would explain a few things.

Liger0, 2020-12-16:
> 
> 
> I can't reproduce this (also on Win10 x64).
> 
> Idea: Do you have 'screen scaling' on? It _should_ still work _with_ screen scaling on, but it would explain a few things.

windows font is set to 125%

konskarm, 2020-12-17:
I think you are hitting this https://github.com/Ultimaker/Cura/issues/8568, but instead of increasing a lot in size with little arrow movement, yours has the opposite effect and cannot increase much. Does this happen to the model the support blocker is attached, too? Did you by any chance scale down that model?

Liger0, 2020-12-17:
https://youtu.be/jwRM8yP80hY

If you try to scale the model, this happens. 
What I noticed is that, if you take the tip of the arrow and try to resize versus the origin point, once you surpass the origin and go to the left it will oversize instead of making it smaller, changing the resizing type, while if you start to drag the center and go to the left, so the opposite way of the axis, you can make it as smaller as you want. 
I am not sure if it is a wanted behaviour.
Also, if you increase it in size of any amount, making it smaller is very hard.

Ghostkeeper, 2020-12-22:
I've seen this issue before. I think I've found the reproduction steps now but it's completely different from what the title implies. My reproduction steps are:

* Load any model into the scene (such as a cube).
* Rotate the model (e.g. 45 degrees around the Z axis).
* Scale the model up significantly using the tool handles.
* Try scaling the model down using the tool handles. This step will partially fail. The effect of dragging the tool handles is greatly reduced.

Ghostkeeper, 2020-12-23:
Apparently Github has a feature now to upload clips, so here goes:

https://user-images.githubusercontent.com/2448634/102983967-27d39080-450d-11eb-98b2-50addcb4593b.mp4

nallath, 2021-10-04:
So it took a while, but we finally got to fixing this: https://github.com/Ultimaker/Uranium/pull/742