Ultimaker/Cura #8922: THICKNESS Changes when Using Lay Flat command "Twice" for full lay on bed , even the Slicers Last layer is not complete

Ruweshed, 2020-12-10:
[image]
[image]
[image]

You can see the THICKNESS actually is 5 mm ,, then after using Lay Flat to lay it on bed it becomes 6.4   beside that the LAST layer is rubbish.... its halfs or quarters...  Hope it get fixed

nallath, 2020-12-10:
Can you also share the model?

Ruweshed, 2020-12-11:
That's the Model  - just click Lay Flat twice ... and check the THICKNESS ,, goes from 5 mm to 6.4 mm  

also Slice it after laying it on bed.... and see how bad are the last layer...

[Ruweshed 3.zip](https://github.com/Ultimaker/Cura/files/5677225/Ruweshed.3.zip)

fieldOfView, 2020-12-11:
The two texts are not equally thick. The layflat tool naively tries to find a way to have tree points touch the buildplate. Apparently the second time the layflat tool is used, it finds a solution where the three points make the model rotate a little.

Compare it to a wooden board that is not quite flat. Depending on where you put your weight when you stand on it, it will have slightly different orientations.

Cura is not doing anything wrong here. The model does not have a nice flat side so there are multiple stable or not so stable orientations where three points of the model touch the buildplate.

closed 2020-12-12 by Ruweshed