Ultimaker/Cura #8923: Scrollwheel support for layer selector in preview.

raf1312, 2020-12-10:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Implement ability to use scrollwheel when mouse hover over the layer number in previwe

Dear developers,

would be nice if it where possible to change the layer in preview by using the mousewheel when hovering over the layer number.
It is difficult to hit the right number per drag of the point with the mouse. 
Using the arrow keys is quite nice, but when you do something with the mouse with the right hand it would be handy to just use the scrollwheel.

Thank you in advance. =)

Timocop, 2020-12-18:
Yes, please! This was bugging me from the beginning.

StuSerious, 2020-12-18:
Oh god I would love to see that! 
If I may, I would also like to suggest that PgUp/PgDwn be bound to previous/next layer or +1/-1 layer as a _very_ valuable feature.
Removes the 'placing-your-cursor-on-the-slider-and-use-scroll-wheel' part and seems quicker both to use and implement.
I often find myself going through layers and that would make my life so much easier!