Ultimaker/Cura #8927: Wont save g code (15.04.06)

JohnWCA, 2020-12-10:
**Application version**
(The version of the application this issue occurs with.)
15.04.06
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows
**Printer**
(Which printer was selected in Cura?)
Sunlu S8
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.) 
Tried downloading files from sd card as my sunlu printer instructed and im unable to save those files. I need to save the files with the right settings to do my first print and im unable to.
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-12-10:
Version 15.04 is no longer supported (and has never been supported here). It is a 5 year old piece of software. You may try the Ultimaker Community forum, or better yet contact the manufacturer of your 3d printer.

Also, you did not fill in the issue template, so there really is nothing we can do to help you.

JohnWCA, 2020-12-12:
Well, it had to of been supported there at some point because the manufacturers help link sent me to you guys and the issue that I wrote was pretty cut and dry. I tried the community forum before sending out the first help request with no solution to my problem. I ended up figuring it out myself, thankfully. But, you’re right about one thing, next time I’ll just get in contact with the manufacturer. Thank you for your help.

jellespijker, 2020-12-13:
@fieldOfView is correct we nog longer support 15.04
As I gather from your comment you figured out what went wrong:

> I ended up figuring it out myself, thankfully

I will now close this issue

closed 2020-12-13 by jellespijker