Ultimaker/Cura #8928: Pre-heat button on Extruder1 conflicts with tooltip

edolis, 2020-12-11:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
Custom, 3 extruders

**Reproduction steps**
1. select monitor tab
2. try to set temperature and preheat extruder 1. The tooltip explaining the functionality keeps blinking on and off over the temperature and pre-heat button. This is conflicting with attempts to click the underlying Pre-heat button. Need to click repeatedly to try to get the lucky time windows the tooltip is off

This seems to be linked to the fact that there are multiple extruders in the panel. No issues with extruder 2 and 3

fieldOfView, 2020-12-11:
A screenshot is really helpful

edolis, 2020-12-13:
[image]
Here you go. The grayed area is the tooltip blinking. If you try to click on Preheat, nothing happens

Ghostkeeper, 2020-12-18:
Those two extruders are also expected to be positioned besides each other rather than underneath each other. That may have something to do with this problem.

edolis, 2020-12-18:
well, I don't know why this screenshot came like that. First time it happened, the extruders were two beside and one lower, but the issue was exactly the same.

Ghostkeeper, 2020-12-23:
Oh, so then the issue is not caused by the layout. Good to know.

closed 2024-12-08 by github-actions[bot]