Ultimaker/Cura #8929: AVR ISP handling crashes Marlin

StefanBruens, 2020-12-11:
https://github.com/Ultimaker/Cura/blob/b49d5ab9e91279ccc89dd14a0f5bf6a4887b51d6/plugins/USBPrinting/AutoDetectBaudJob.py#L34

https://github.com/Ultimaker/Cura/blob/b49d5ab9e91279ccc89dd14a0f5bf6a4887b51d6/plugins/USBPrinting/avr_isp/stk500v2.py#L47

The magic sequence sent to the printer makes Marlin crash.

1. This is a serious problem when Marlin fails to reset itself. I had it hang while the Extruder was using a high-duty PWM cycle, after a short time it started burning filament. (This was with Marlin 2.0.x from February 2020, 2.0.7.2 seems to properly reset at least).

2. Trying to use some AVR ISP code unconditionally, when more and more printers become ARM based seems strange.

nallath, 2020-12-14:
>     1. This is a serious problem when Marlin fails to reset itself. I had it hang while the Extruder was using a high-duty PWM cycle, after a short time it started burning filament. (This was with Marlin 2.0.x from February 2020, 2.0.7.2 seems to properly reset at least).

Hmm, that is odd. It's essentially the same code that we've been using since Cura 14.00 (or perhaps even longer). This makes me lean a bit towards the "But Marlin shouldn't crash on this" side of the argument. 

>     2. Trying to use some AVR ISP code unconditionally, when more and more printers become ARM based seems strange.

True. Although I do want to add that USB printing never really got our focus and people actually switching over to ARM should even decrease the need of it further (since the machine should just handle this in the first place)

StefanBruens, 2020-12-14:
> Hmm, that is odd. It's essentially the same code that we've been using since Cura 14.00 (or perhaps even longer). This makes me lean a bit towards the "But Marlin shouldn't crash on this" side of [...]

I never had problems with the stock AVR based board with Marlin 1.1, but then I switched to an ARM (STM32F1) based one with Marlin 2.0.x, and since then it crashes.

> > ```
> > 2. Trying to use some AVR ISP code unconditionally, when more and more printers become ARM based seems strange.
> > ```
> 
> True. Although I do want to add that USB printing never really got our focus

Problem is, the crash even happens when not actually using USB printing, just connecting the printer is enough to trigger the USB probing. I.e. when I try to directly save to the printer SD card using Marlins Mass Storage mode from Cura, it crashes  as soon as the printer is connected.

> and people actually switching over to ARM should even decrease the need of it further (since the machine should just handle this in the first place)

Sorry, I don't understand this part of the sentence. The need for what?

nallath, 2020-12-15:
> > and people actually switching over to ARM should even decrease the need of it further (since the machine should just handle this in the first place)
> 
> Sorry, I don't understand this part of the sentence. The need for what?

Directly streaming the g-code from a computer (instead of using an SD card or the like)

Ghostkeeper, 2020-12-18:
The header there is a perfectly valid sequence to start the STK500 communication. It enters ISP programming mode, uses a programming lock, enters PP programming mode and then gets a parameter, validates that against 0x1000. I'm not sure what that parameter is, but it seems perfectly fine to me.

I also think this is a bug that should be fixed in Marlin rather than in Cura. Marlin shouldn't crash on a perfectly valid STK500v2 protocol.

closed 2020-12-18 by Ghostkeeper

StefanBruens, 2020-12-18:
You are sending an arbitrary sequence to a random serial port. This may be any device, a braille line, some measurement device... You are voluntary risking to damage hardware, and there is no way for the user to stop Cura from doing that.

There is no configuration, no USB PID/VID filtering, nothing.

StefanBruens, 2020-12-19:
@Ghostkeeper - please answer this. Why is Cura sending arbitrary byte sequences to any serial port?

Ghostkeeper, 2020-12-23:
That is indeed considered a bug: https://github.com/Ultimaker/Cura/issues/5207

This bug is about the STK500 handshake apparently crashing Marlin, not about Cura connecting to devices which are not 3D printers. Otherwise I'd have closed this issue as a duplicate rather than as a Won't Fix/Do.