Ultimaker/Cura #8931: Nozzle temp. falls to 0 3 layers before end

smartkidag, 2020-12-11:
**Application version**
4.8.0

**Platform**
Windows

**Printer**
Custom Printer

**Reproduction steps**
1. Created a custom printer with 2 extruders.
2. Set desired coniguration (nozzel size, temperatures etc)
3. When I sliced it and printed it, the temp would go down to 0 right at the end and the extruder would stop (to avoid cold extrusion).

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
The slicer puts in a M104 T0 S0 and the nozzle temp falls before during the print

**Expected results**
The temp should stay stable at set value until the end

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
This is a custom ender 3. Never had such trouble before. Sure, I can manually remove it but I really don't want to read through the Gcode everytime (or find it everytime).

[CFFFP_xyzDualColorHollowCalibrationCube-letters.zip](https://github.com/Ultimaker/Cura/files/5681683/CFFFP_xyzDualColorHollowCalibrationCube-letters.zip)

nallath, 2020-12-14:
This is expected behaviour; The nozzle is done printing (and is thus set to 0)

closed 2020-12-14 by nallath

fieldOfView, 2020-12-14:
Cura does not know that you have two extruders sharing the same heater. You need to turn on the "machine_extruders_share_heater", either in your custom printer definition, or using the Printer Settings plugin.