Ultimaker/Cura #8932: [4.8] Cura temporarily freezes when exporting a file with post elaboration scripts

Liger0, 2020-12-11:
**Application version**
4.8

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Import any stl, even a stringign tower which is very small
2. Apply a change at z retraction
3. Try to export the file

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
The application totally freezes for about a minute after you select the folder (I tried different drives)

**Expected results**
No hang.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/5681923/cura.zip)

**Additional information**
I didn't have this issue with older versions. If you remove ChangeAtZ it returns to be smooth

evtrados, 2020-12-16:
Hello @Liger0 ,

Thanks for reporting this. I could, however, not reproduce it using the reproduction steps you mentioned above. I tried with several printers, selecting several drives and using several post processing scripts, but with no luck. The only thing I can do is ask another colleague if this scenario is reproducible for him/her.

Liger0, 2020-12-16:
Did you try switching to save as file rather than a direct automatical drive detection?

Ghostkeeper, 2020-12-18:
The issue here is with the performance of Python scripts, and the CPython GIL. This is a duplicate of https://github.com/Ultimaker/Cura/issues/6468.

It happens a lot with some of the scripts (post-stretch is particularly bad with performance) and large g-code files.

closed 2020-12-18 by Ghostkeeper