Ultimaker/Cura #8934: Print overlap in one at a time mode renders under the plate

billy1380, 2020-12-13:
**Application version**
4.8.0
**Platform**
Mac OS X (details in screen shot)
[image]

**Printer**
Prusa MK3

**Reproduction steps**
- Set print one at a time mode
- Load a model
- add multiples of the model

**Screenshot(s)**
[image]

**Actual results**
The print overlap shadow is rendered under the plate

**Expected results**
The print overlap shadow (rectangle) renders under the plat so it is only visible when the object is moved away from the plate

billy1380, 2020-12-13:
Seems to be a problem when the raft option is enabled. When it is disabled the shadow comes back.

[image]

rburema, 2020-12-16:
Thanks for the report. We've created internal ticket CURA-7930 to address this. It might be a while until we get to it (if at all) though!

billy1380, 2020-12-16:
@rburema I understand... it is not critical.

As a workaround I switch off build plate adhesion until before I generate gcode.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-13 by github-actions[bot]