Ultimaker/Cura #8935: [Feature request]]

userosos, 2020-12-13:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

userosos, 2020-12-16:
If i select an profile after it i select material (spool) Profile drop to default - fine. It is bug?

konskarm, 2020-12-17:
If you want to add filament cost/weight information, do the following:

1. Open the Materials tab in the preferences (Ctrl+K)
2. Select the material you want (e.g. Generic PLA)
3. Press Duplicate (this will create a duplicate of the material you had selected)
4. In the duplicated material, press the "Unlink Material" button
5. Fill in the Filament Cost and Filament Weight values in your new material

[image]

> If I set a custom material, what are the implications for Print settings/Profile?

Using the duplicated material should work exactly like using the original one you duplicated, so it shouldn't affect your print quality at all.

> I would like to be able to display different materials to estimate the cost of printing ONLY.

Once you slice, you can see the cost of each material by hovering over the information icon, down right:
[image]

> Also add, please, the ability to view the cost of printing a model separately, if the table there are more than one.

This is a feature request. I'll put it up for discussion.

> If i select an profile after it i select material (spool) Profile drop to default - fine. It is bug?

I don't really understand that. Can you provide detailed reproduction steps?

konskarm, 2020-12-17:
We discussed the feature request about pre-calculating the cost of each object, and we decided to reject it, considering that it requires quite a lot of work to be done in our engine in order to achieve it. So the work needed doesn't justify the benefit of the feature itself.
As a workaround, you can see a good estimate of the cost simply by slicing the specific model alone on the buildplate.

**Note:** I renamed the feature request title so that other users can find it and refer to it if needed. In case you find another bug, please create a separate ticket for it.

closed 2023-07-13 by github-actions[bot]