Ultimaker/Cura #8936: [4.8.0] Crashing at 3rd step when getting started

Hema2-official, 2020-12-13:
**Application version**
4.8.0

**Platform**
OS: Windows 10 Pro 20H2
GPU: Intel HD Graphics 3000 (Integrated)

**Printer**
*No printer selected, I didn't even reached printer selection*

**Reproduction steps**
1. Opened Cura, and it started normally.
2. Clicked on "Get Started"
3. Clicked on "Agree" (User Agreement)
4. Clicked on "Next" (What's new in Ultimaker Cura)
5. Clicked on "Next" again (Help us to improve Ultimaker Cura, *last click*)
6. It crashed (Just closed and left me alone)

*Cura right before crashing:*
[image]

**Actual results**
It just crashed. That's it.

**Expected results**
I expected I can continue the Getting Started part, but... yeah

**Project file**
*No project file, nothing was opened*

**Log files**
***cura.log** and **stderr.log***
[cura.log](https://github.com/Ultimaker/Cura/files/5684645/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/5684644/stderr.log)

nallath, 2020-12-14:
I think this might be due to some graphical incompatibility. Have you tried upgrading your graphic card drivers?

Hema2-official, 2020-12-14:
Yeah, that could be it.
My graphics driver is always the latest, but this is the newest "*doesn't support everything*" integrated GPU. If you know what I mean. It doesn't support DX11 so... (It can't even run Fortnite or PUBG Lite!)
Thank you for your comment!

closed 2020-12-14 by Hema2-official