Ultimaker/Cura #8937: App is freezing after last update

equipment66, 2020-12-13:
Hi.

Ultimaker Cure v4.8.0 continuously freezing during operation. Situation was noted after last update of MKS plugin. Freezing appears continuously, on every operation inside UI - manipulating with 3d objects, switching windows or using menu. Freezes varies from 1 to  5 -10 seconds.

**Application version: 4.8.0**

**System: Win10 x64**

**Printer: It takes to mach time to find it with freezing app :(**
(Which printer was selected in Cura?)

nallath, 2020-12-14:
If I understand you correctly, the issue only occurred after upgrading the MKS plugin? If that is the case, please contact the developer of that plugin.

equipment66, 2020-12-14:
> Closed  #8937.

Hi.
  

  
Not absolutely correct. It was noted after last upgrade of MKS. But many guys who are working with FB Ghost from Telegram channels complaints about slowing down Cura right now.

nallath, 2020-12-14:
Is the issue still there if the plugin is not installed?

equipment66, 2020-12-14:
1) will try uninstall it this evening;
  
2) will make short screen video for you before.
  

  
Now I'll go offline for about 2 hours.

equipment66, 2020-12-14:
Good Afternoon.

As promised, some videos for you.
Btw, it works much better when screen recorder is active. Mostly likely because priority of processes is changed. But, anyway, you can see freezes on the screen.

https://youtu.be/CnUXljG7Ik0

https://youtu.be/Wkd5ZYdyaU4

https://youtu.be/4Eiha71UXNs

Btw, how to uninstall MKS correctly? Simply delete it from plugins folder?

Geros dienos / Have a nice day,

Vilius Grinkevičius
GSM: +370 660 05003

From: Jaime van Kessel
Sent: 2020 m. gruodžio 14 d., pirmadienis 15:38
To: Ultimaker/Cura
Cc: equipment66; Author
Subject: Re: [Ultimaker/Cura] App is freezing after last update (#8937)

Reopened #8937.

closed 2020-12-29 by no-response[bot]