Ultimaker/Cura #8938: Cura could not be used via touchscreen with Windows 10 X64 tablet

NilsRo, 2020-12-13:
**Application version**
4.8.0

**Platform**
Windows 10 X64 installed on a tablet Dell 5290

**Printer**
Anycubic Mega S

**Reproduction steps**
1. Try to select a menu via touchscreen

**Screenshot(s)**
Video Doku: https://youtu.be/hNVj2YtjU-M

**Actual results**
Could not select the menu

**Expected results**
Menus should be possible to select via touchscreen

**Project file**
-

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5685027/cura.log)

**Additional information**

evtrados, 2020-12-16:
Hi @NilsRo,

Thanks for reporting this. Unfortunately this is a known issue and is still on our backlog. There is some re-prioritization going on, so I hope it gets the attention it needs in order to have it fixed for the next release or the one after that.

evtrados, 2020-12-16:
Developers, see ticket number CURA-6216.

Ghostkeeper, 2020-12-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/2515.

closed 2020-12-18 by Ghostkeeper

NilsRo, 2020-12-19:
@Ghostkeeper
Hi. You closed the ticket but the linked ticket does not really match this one. The related ticket is about **better** touchscreen support (that would be really nice sitting next to the printer :-) ) but anyhow the ticket here is that cura **cannot** be used with a touchscreen. So the content of the ticket is different. It seems a bug of the window framework used or something like this.

Ghostkeeper, 2020-12-22:
The linked ticket is about touchscreen support. Currently we don't support touchscreen devices at all.

It's indeed related to the windowing framework, Qt. But it's not a bug in Qt, just how we're using it. Qt has features for touch support, such as the `MultiPointTouchArea` or the `Flickable` elements. We're using `Flickable` here and there by accident through using Qt's built-in scroll views, but that's not done consistently. And navigating the 3D scene is completely impossible since we're just asking Qt for the mouse coordinates there and tracking when the mouse buttons get pressed, which doesn't trigger when you're using a touch screen.

To implement this, camera navigation and 3D scene interaction need to be reworked. We'd need to fix scrolling of elements here and there. And there are some operations that we currently use the right mouse button for (e.g. the context menu containing "Reset all model positions") which would need to get different UX flows.

fieldOfView, 2020-12-22:
> But it's not a bug in Qt, just how we're using it.

The problem in this issue is though. There should not have to be a need to do anything special with the handling of touch messages to open an application menu.

NilsRo, 2020-12-22:
Yes. Even without special support the application menus and everything else should behave like using a mouse if Touch is not supported. But the application menus goes crazy. That is what I do not understand and which makes Cura unusable in tablet mode. I do not need anything special (but there would be much optimizations possible like you listed), but it would be nice to take the tablet and switch from my homeoffice to the printer room.

NilsRo, 2021-02-21:
@Ghostkeeper Any news if basic click support will be added (no special support, only that I can click on a menu)?

Ghostkeeper, 2021-03-01:
The original that this ticket is a duplicate of is still in our backlog, meaning that Ultimaker has plans to eventually build it and a priority set for it. The priority for it isn't particularly high (some 200 tickets before it) so I wouldn't expect it at least for a year.