Ultimaker/Cura #8940: [4.8] Change at Height - Speed affects entire print

Smoghog, 2020-12-13:
**Application version**
4.8

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Set a Change at Z height to reduce my print speed to 50% at Layer X in my design

**Actual results**
The print starts at 50% Feed Rate and stays there, also does not revert back to 100% when finished, so the printer is still at 50% when starting an unrelated print next, which was particularly odd.

**Expected results**
My hope was that the project would print at normal speed until Layer X, then reduce to 50%, and the printer would go back to default values at the end of the print.

I recognize that the 'change at height' post processing features is still in its experimental phase, and might not work entirely as intended. I wanted to submit my results in case it helps improve the feature, not to complain! :D Thank you!

GregValiant, 2020-12-15:
I've been caught with this before.  In my case I had made a manual adjustment on the LCD and there was no reset command either at the end of the current file, or in the start of the next one.  So I added it to my End Gcode in the machine settings.
M221 S100
M220 S100
Now it always resets at the end of a print and the next print starts at 100% flow and feed.

Smoghog, 2020-12-15:
> I've been caught with this before. In my case I had made a manual adjustment on the LCD and there was no reset command either at the end of the current file, or in the start of the next one. So I [...]

Good call! I hadn't thought to do that, but I'm going to for sure. I just noticed the problem happened with Flow tonight as well. It seems that it doesn't happen when you check 'change pr int speed', but it does happen if you just check 'change speed', so I'm using it for print speed adjustment only at this time.

Ghostkeeper, 2020-12-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/8574. GregValiant here also reported the issue before :heart:

closed 2020-12-18 by Ghostkeeper