Ultimaker/Cura #8941: Acceleration Settings for Bridging

dlinders10, 2020-12-13:
I would like for there to be acceleration settings for bridging. I like to use lower custom acceleration settings for my prints so I can get detailed outer walls but the low acceleration messes up bridging. If I use higher acceleration values, bridging is improved but the rest of print isn't as good. This feature is in Prusa slicer and works very well enabling low accelerations where needed and high acceleration which is necessary for good bridges.

smartavionics, 2020-12-13:
See https://github.com/smartavionics/Cura/releases/tag/20201121 for a Cura variant that provides accel/jerk settings for bridge walls and skins.

Ghostkeeper, 2020-12-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/4955.

closed 2020-12-18 by Ghostkeeper