Ultimaker/Cura #8942: No downside holes on Cones

RoiWater, 2020-12-14:
**Application version**
4.8.0

**Platform**
Windows 10 64bit Version 1909 (Build 1863.1198)

**Printer**
Creality Ender-3 Pro

**Reproduction steps**
1.  Open Conic Stl-File with holes inside (from the downside) like DS18B20 Solar Shield (Thing https://www.thingiverse.com/thing:2553127 DS18B220_Solar_ShieldV2_-_Cap.stl)
2. Printout oder verifiy via preview that there is no hole generated in the layers
3. Open same file in an alternative Slicer like the prusa slicer
4. Printout oder verifiy via preview that there is a hole generated in the layers

**Screenshot(s)**
[image]
Slicing with cura 4.8.0

[image]
Slicing with Prusa slicer

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
I wouuld have expected to have the same holes insinde the cone (for attaching the Cap to the base) with Cura as if I would slice the Cap with Prusa or a other slicer

**Project file**
[CE3PRO_DS18B20_Solar_ShieldV2_-_Cap.zip](https://github.com/Ultimaker/Cura/files/5687857/CE3PRO_DS18B20_Solar_ShieldV2_-_Cap.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I created a cone with Fusion 360 with a accoring hole just to verify that the problem is not related to that specific STL file, but the result are the same.

Asterchades, 2020-12-14:
It's working just fine for me.

Check to make sure that you don't have either **Make Overhang Printable** (a known bug causes it to currently fill in some volumes touching the build plate) or **Remove All Holes** (it's supposed to fill everything within the outer-most wall). Both of those can cause the same behaviour you're seeing but neither should be the default.

nallath, 2020-12-14:
Overhang printable is indeed enabled in the profile. So it's working as expected.

closed 2020-12-14 by nallath

RoiWater, 2020-12-16:
Confirmed, that if overhang printable is disabled the effect will not appear. If there is already a bug report for make overhang printable filling some volumes touching the build plate its fine to me to close this bug.

Roi

fieldOfView, 2020-12-16:
> for 'Make Overhang Printable' filling some volumes touching the build plate

Why would there be a report for that? It is exactly what the feature is supposed to do.

Asterchades, 2020-12-16:
Because it can still be printable already without filling in the hole, as would be the case in this model where the overhangs are only 45° or so - even if they happen to build a cap over the build plate.

https://github.com/Ultimaker/CuraEngine/pull/1379

RoiWater, 2020-12-16:
@filedOfView: I would not expected with this option that there is totaly no hole in the object (see screenhots), but would have expected steep overhangs to become shallower overhangs as in the description of the feature. In the example it should be possible to extend the hole into the infill area to obtain such a behaviour.