Ultimaker/Cura #8943: 4.8.0 and 4.4.0 Cura error on start up

Oj00, 2020-12-14:
**Application version**
(The version of the application this issue occurs with.)

4.8.0 and previously 4.4.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Win 10 PRO, Version 1909, GPU: NVIDIA Quadro P2000

**Printer**
(Which printer was selected in Cura?)
Happens before printer selected, printers connected via network are UM3, UM3E and S5

**Reproduction steps**

Each time I start Cura I get the first time set up screen, and error (shown in photo) behind, which I cant access until I've finished the set up. Then I click to reset- which reboots Cura which then needs first set up but no error. 
The next time I start Cura the process repeats itself. I'm on the latest version, and it has continued happening after the update. (UM3, UM3E and S5)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log](https://github.com/Ultimaker/Cura/files/5688081/cura.log) (4.8 version folder)
[cura (1).log](https://github.com/Ultimaker/Cura/files/5688082/cura.1.log) (4.4 version folder)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-12-14:
Are the printers that you are connected to in a single group? 

As far as I can see, it's not happening every boot, so there must be something that causes it. Do you know what actions you've taken in the cases that it causes this issue?

Oj00, 2020-12-14:
All three should be in the same group, they all connect as one. 

This error occurs every other boot, after you open it and get the error, it gives you the option to reset the configuration. Once this is done it will open without the error and allow you to re-add the group of printers. The next time you restart Cura it appears with the same issue as before.

evtrados, 2020-12-15:
Hi @Oj00 ,

This is probably caused because you still have the 4.4 configuration folder and something goes wrong when upgrading from these configurations to the ones in 4.8. Could you maybe rename that folder to something like 4.4-backup and try again? These configuration folders are in the %APPDATA%/cura directory.

Oj00, 2020-12-16:
Thanks, that's resolved it. Not sure why both sets of files were still there, when I installed the new version I click yes to un-install the previous version.

closed 2020-12-16 by Oj00