Ultimaker/Cura #8944: [4.8] Adding ability to select/print with CC 0.6 print core on UM3

marine-3d, 2020-12-14:
Whilst I understand that due to the feeders on the UM3 not being hardened, the CC 0.6 core isn't supported on the UM3.

However, the printer can use the core, works well and as long as people are aware of the risk (Which is clearly stated when purchasing the core), could the core not be added to the dropdown list? Maybe as an additional precaution adding text 'Unsupported' next to the core, similar to what is done with experimental profiles.

The positive of adding this feature would be:
- Whilst there is an option of buying the AA 0.8 core, the combination of slightly faster print times (vs AA 0.4) with a better finish (vs AA 0.8) is a good reason for selecting the CC 0.6 over the AA 0.8.
- Don't have to necessarily print with abrasive materials on the CC 0.6.
- Option to changing the feeders e.g. Bondtech to support abrasive materials.

[image]

nallath, 2020-12-15:
We would also have to re-do the profiles for those cores, which isn't a trivial task (eg; it's not as simple as just copying some profiles). 

Adding some extra warning labels would decrease the number of issues that people would get from it, but experience has taught us that it won't completely remove them. So those things combined do mean that it's not something that we will implement.

closed 2020-12-15 by nallath

marine-3d, 2020-12-16:
I appreciate it might not be a direct copy and paste and the focus is on the newer printers, though it does seem a pity the majority of the work/information is there and Cura/UM3 is being held back.

As you might have seen in the forum, I have copied the profiles, variants and materials from the S3 and currently seems to be working well. Any suggestions on what I should be keeping an eye on in possible differences between the UM3/S3?

Many thanks

Ghostkeeper, 2020-12-18:
The UM3 has different stepper drivers which will not be able to reach the same acceleration as the UMS3.

The UMS3 has more up to date software, which isn't necessarily better for printing but *different*. It should position more accurately in curves, reducing the [aliasing](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/aliasing.md) effect. However in the past it's also taken more processing power for auxiliary tasks which caused buffer underruns, so you could expect to need to increase the Maximum Resolution setting. They made a lot of improvements for the latter issues but as far as I know it doesn't yet exceed the UM3.

The feeder is also different, as you already know. I'd use a bit more conservative retraction settings. In its build, the UMS3 is more akin to a scaled down UMS5.

Those are the three biggest differences as far as the profiles are concerned I think. The cooling hasn't changed as far as I know, and the heating track and nozzle are the same of course (because that's the part you're swapping in).