Ultimaker/Cura #8945: auto scale when loading tiny STLs

noahdudly, 2020-12-14:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

noahdudly, 2020-12-14:
When in STL is very tiny, the most likely cause is that it is in inches not MM, Scaling when loading should generate a option, not just a notification, and default to 2540, because metric is better.

konskarm, 2020-12-15:
There is already a setting in the Preferences that decides whether to scale small objects on loading. Is that what you are looking for?

[image]

Zogar89, 2020-12-15:
> 
> 
> There is already a setting in the Preferences that decides whether to scale small objects on loading. Is that what you are looking for?
> 
> [image]

That doesn't work for inch models. I try some.

nallath, 2020-12-16:
The problem with inch models is that there is an overlap between models that should be really small and models that are small because of a inch vs mm mismatch.

The scale extremely small models is mostly intended to fix the meter vs mm issue (since then it is clear that it's going on).

As for the inch conversion, i've made a quick plugin that makes it a bit simpler: https://marketplace.ultimaker.com/app/cura/plugins/nallath/BarbarianPlugin

closed 2020-12-16 by nallath