Ultimaker/Cura #8946: Possible memory leak for 4.8.0 - Windows Version

nschoe, 2020-12-14:
**Application version**
4.8.0

**Platform**
Windows 10 - Intel i7-8700 - 32GB RAM

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. Start Cura
2. Let it open in the background (no slicing needed, nothing)

**Screenshot(s)**
N/A

**Actual results**
Memory consumption gradually increases over time, and at some point Cura becomes unusable because very slow.
Some measurements I made:
- 09h22 a.m. 283.2MB
- 11h44 a.m. 439.4MB
- 03h24 p.m. : 691.3MB

**Expected results**
Steady memory consumption

**Project file**
N/A

**Log file**
N/A

**Additional information**
This happens whether or not I previously sliced a file and/or launched a print.
Every time I open Cura it's the same.
Note that I have two Ultimaker S5s on the network, and "paired" in Cura.
It happens whether or not they are printing.

Hope it helps!

evtrados, 2020-12-16:
Hello @nschoe ,

Thanks for reporting this. I will also have a look myself if I can reproduce it and then bring it up with the rest of the team.

evtrados, 2020-12-22:
Sorry for the somewhat late response on my findings. So, I've tried this on two laptops and one PC that run Windows 10. On all three occasions, I could only see consumption up to 310MB. In the beginning (until Cura becomes idle) the consumption is increased and sometimes reached 360MB, but then it gradually stabilizes and on most occasions it remained below 300MB. I will, in any case, brig it up with the rest of the team to see what they think.

Ghostkeeper, 2020-12-24:
I've reproduced this issue on my Windows installation using the 4.8 release. My measurements of memory usage:

At 19:40 I started Cura and it used 285.4MB memory.
At 20:40 it used 322.0MB memory.
At 21:50 it used 335.2MB memory.
At 23:20 it used 410.6MB memory.

This was with one Ultimaker 3 on the network that I connected to Cura. It's a fresh installation, the only thing I did was to add that Ultimaker 3 to Cura and leave it connected while I was playing Cyberpunk 2077.

There are multiple variables that were different between my test and Evtrados' test. But the one we suspect might be the issue is Zeroconf. I have a printer on my LAN, and Evtrados didn't. We've seen memory leaks in Zeroconf before and contributed a fix to them back then, but maybe something got back in.

nschoe, 2020-12-24:
Thanks @evtrados and @Ghostkeeper for taking interest in this and make your own tests.
I can confirm the issue is still happening with me, which incidentally is a big issue because it forces me to close Cura after I launched my prints, and sometimes when the print fails and I need to tweak a few features, I have to re-set everything :/

Reading @Ghostkeeper 's comments, I've checked: I *also* have some printers on the network. Don't know much about Zero Conf, but when I search them on the network, I find them. So I suppose it's related?

Thanks again for your time and tests, let me know if I can be of any additional help.

Merry x-mas to you all :)

Ghostkeeper, 2020-12-24:
Zeroconf is automatic discovery of devices on the network. As long as the UM3NetworkPrinting plug-in is enabled (which is used to connect to all Ultimaker printers, not just the UM3) then Zeroconf will be running. You mentioned that you had 2 Ultimaker S5's in your network both connected to Cura, which is what led us to this thought. So thank you for including that detail ;)

This issue is now on our backlog because the developers think it's important enough to get short-term attention. It's up to our quality assurance to prioritise it further. I expect that we'll get around to it in 2 or 3 months or so (if QA agrees with the priority that I would give it).

nschoe, 2020-12-24:
Okay I see.

Well I am remote-working right now, I also have Cura 4.8.0 on Windows 10 installed on my home computer.
I just started Cura and I'll let you know if I also have this apparent memory leak.

I don't have Ultimaker printers at home, but i do have an Ender 5 Pro with a RPi 3 running Octoprint, on the same network (connected via Ethernet).
Would Zero Conf be involved here?

Ghostkeeper, 2020-12-28:
Yeah as far as I know the Cura Octoprint plug-in also runs Zeroconf.

Floyddotnet, 2021-04-06:
I have the same problem and can confirm that disabling the UM3NetworkPrinting plugin solved the problem.

0d 00h Cura start - 121 MB memory
0d 17h Cura use 600 MB memory
1d 03h Cura use 1100 MB memory

after disabling UM3NetworkPrinting :

0d 00h Cura start - 121 MB memory
1d 03h Cura use 125 MB memory

fieldOfView, 2021-04-07:
The OctoPrint Connection plugin includes a different (newer) version of the ZeroConf module; 0.25.1. Cura seems to include 0.24.1. That may explain the difference.

fieldOfView, 2021-04-07:
At least one memory leak was resolved between 0.24.1 and 0.25.1:
https://github.com/jstasiak/python-zeroconf/commit/eac53f45bddb8d3d559b1d4672a926b746435771

nallath, 2021-07-20:
We have updated to the latest & greatest of Zeroconf (0.31.0) and the issue still persists. So whatever is causing it, it's not fixed by updating zeroconf...

closed 2024-12-08 by github-actions[bot]