Ultimaker/Cura #8948: blobs after modify gcode filament change

jmerc77, 2020-12-15:
cura 4.8.0
(The version of the application this issue occurs with.)

windows 10 64bit

16GB ram 
intel i7

CR-10S (marlin 2.0.6.1)

**Reproduction steps**
1. sliced with a filament change modifier
2. printed and wait for pause
3. switch filament using the on-screen steps (including wiping nozzle)
4. wait for start.

Screenshot and project file attached.
[image]

[CCR10S_digits.zip](https://github.com/Ultimaker/Cura/files/5692311/CCR10S_digits.zip)

**Actual results**
squirt. a blob appears.

**Expected results**
no blobs, please.

**Additional information**
it seems to be an un-retract issue. there should not be an un-retract after filament change.

nallath, 2020-12-15:
What is a filament change modifier?

jmerc77, 2020-12-15:
in cura under extensions->post processing->modify gcode. click add script and select filament change.

kejones62, 2020-12-17:
I'm seeing the same thing.  I looked at the spec for the M600 gcode which is inserted by this feature, and I see that the "initial retraction" number is supposed to be negative; by default Cura inserts a positive number (and if a user-specified negative number is used, Cura does not print the initial retraction spec at all).  This is causing the printer to spit a blob, rather than retract the filament. Using an editor to set the retraction to a negative number in gcode results in the proper behavior by my printer (Ender 3).

While this bug is being fixed, I recommend a review of the M600 parameters and their use by the code.  The "later retraction" option is being used to set the M600 'load length' - a different setting - and it also has the negative number issue noted above. 

Finally, as an enhancement, it would be nice to be able to set ANY of the M600 parameters using the 'filament change' feature, rather than just the 4 seen (initial retraction, final retraction, X, and Y).

jmerc77, 2020-12-17:
pre retract will work to counter the unretract occurring at the cost of a few seconds of time but time is not a big deal in this case.

nallath, 2020-12-18:
> I looked at the spec for the M600 gcode which is inserted by this feature

What spec are you looking at? The one I'm looking at (the one from the RepRap project) doesn't mention this. I'd also expect the number to be positive since a negative retract is a double negative.

kejones62, 2020-12-18:
I use Marlin, and I'm looking at something
https://marlinfw.org/docs/gcode/M600.html, for example.  I agree using a
negative sign is silly, but I can confirm that on my printer (ender 3 with
Marlin firmware) negative retract does the right thing, and positive leaves
a blob.

Ghostkeeper, 2020-12-18:
This has been changed back and forth a number of times:
* https://github.com/Ultimaker/Cura/commit/c7b146c9fc3bc287b456e9eea5b17aad4b689086
* https://github.com/Ultimaker/Cura/commit/296b74b03244ee4ca42c934c8feb43c6c69bb76c
* https://github.com/Ultimaker/Cura/commit/5592d4fa602835536ea7b57778c7fd744294a182

We kept changing it since the Ultimaker printers don't support this g-code so we can't test either way. However when looking at the history of that minus sign there, I must conclude that different firmware interprets this differently.

Looking at documentation, the [RepRap wiki](https://reprap.org/wiki/G-code#M600:_Filament_change_pause) sort of words it ambiguously. The [Marlin Documentation](https://marlinfw.org/docs/gcode/M600.html) clearly says it should be negative, however the [actual Marlin code](https://github.com/MarlinFirmware/Marlin/blob/5cf6c44f47737bdf93df6ae7c835fe2d54d0675d/Marlin/Marlin_main.cpp#L10993) wraps this value in `ABS()` and so whether it's negative or not shouldn't matter to Marlin.

See also the previous bug report about this: https://github.com/Ultimaker/Cura/issues/7500

closed 2020-12-18 by Ghostkeeper

eGit, 2020-12-28:
@kejones62 I think I have the same issue with my Ender3. Just to be 100% sure we are on the same page how you manually changed your G-Code, please paste the M600 G-Code line before and after you manually edited it.

Is it something like this:
Before (generated by Cura 4.8):       M600 E30.00 L300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
After manual editing of the G-Code: M600 E-30.00 L-300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin

Is this about right?

https://marlinfw.org/docs/gcode/M600.html

The Marlin docs should mention if you have to add the "Minus" sign to the numbers in [E<pos>] and [L<pos>] or if you can leave it out and it interprets the number as "negative". This is confusing. What I glanced from other comments in this issue, it seems the number should be positive but the printer should interpret this correctly as "negative", but some printers like the Ender3 does not. Correct?

kejones62, 2020-12-28:
> numbers in [E] and [L] or if you can leave it out and it interprets the

First, I am not an expert in gcode, Marlin, nor the Ender, so I'm only
reporting what I observe.
I removed the 'L' parameter completely when I edited, because I just wanted
to get the print done and not experiment with parameters.  The 'L' should
be used for reloading the filament, which isn't necessary if you do it
manually.  I'm not sure why Cura includes  'L', but doesn't bother to
include 'U', which is the unload retraction.  Anyway,  I edited the code by
removing the 'L' parameter and changing the sign of the 'E' parameter,
thusly:
M600 E-30.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
You do want some retraction (-30), since this prevents stringing while the
head is moving for filament change.

eGit, 2020-12-28:
Thanks. I just tried with the manual edit earlier and it worked. I agree, next time I will remove the 'L' parameter. Maybe I will increase the retraction ('E') to -100 or even higher. When I do a filament change manually, I quickly pull out the filament. This always worked well. With the automatic E600 retraction I often get a bigger blob then usual at the filament end. I think it doesn't retract quickly enough. This results in problems when I want to fully pull out the filament.

Otherwise it works now very well. Probably in Cura 4.9 they will change it back again ;)

Ghostkeeper, 2020-12-28:
We're not likely to change this again because previous change requests have pointed out that there is just no consensus about how this g-code works.

eGit, 2020-12-28:
On the other hand 90% of printers probably are a handful of models. Ender3, etc.
It should be fairly easy to check for which printer this G-Code is written out and write out the G-Code accordingly. I am pretty sure Cura already does this in many other cases.

Ghostkeeper, 2020-12-30:
Cura does that depending on the g-code flavour sometimes. However this doesn't seem to be linked to g-code flavours. MonoPrice and Ender both use the Marlin/Sprinter g-code flavour.

Wob76, 2021-03-22:
Sorry to re-open this but I have been trying to work out why I get this on some prints and not others. I can see in the g-code why it has happened, on a problem print there is a retract before and then a unretract after the inserted M600, since the M600 process primes the nozzle we don't actually need the unretract on return, I would also say we don't need the retract since M600 has a retract built in, I am considering setting M600 to 0 to see if it helps, but it won't change the unretract that Cura is adding.

What I can't work out is why the retract is there on some prints and not others with the same retract settings in Cura. All I did for the tests below is change the object loaded, It seems to just be related to the object properties. The M600 plugin doesn't have the smarts to account for these retract\unretracts, it would be nice if Cura could get some native M600 support.

Test Printer: Ender 3 Pro with SKR board, direct drive modification and custom firmware build, M600 enabled.
M600 Firmware Settings of note (these are the defaults)
#define PAUSE_PARK_RETRACT_LENGTH            2
#define ADVANCED_PAUSE_RESUME_PRIME          0

Retract Settings in Cura;
Retraction Enabled
Distance 2mm
Speed 25mm/s
Retract Before Outer Wall - enabled
Retract at Layer Change - disabled

Problem Print (Note the 2mm retract before the NONMESH and the unretract at the start of the WALL-INNER
G1 F1500 X161.465 Y112.816 E4122.22148
G0 F9000 X161.441 Y112.817
G1 F1500 E4120.22148
;MESH:NONMESH
G0 F300 X161.441 Y112.817 Z4
G0 F9000 X160.101 Y113.762
G0 X105.231 Y159.745
;TIME_ELAPSED:6008.619906
M600 Y10.00 ; Generated by FilamentChange plugin
;LAYER:19
;TYPE:WALL-INNER
;MESH:Fusion Overload Base.stl
G1 F1500 E4122.22148
G1 X105.234 Y159.729 E4122.22202

And on a test Cube we don't see it;
G1 X120.099 Y108.927 E419.30982
G1 X120.249 Y108.938 E419.31477
;MESH:NONMESH
G0 F300 X120.249 Y108.938 Z4
G0 F9000 X120.902 Y109.928
G0 X123.889 Y109.928
G0 X126.5 Y108.5
;TIME_ELAPSED:814.373645
M600 Y10.00 ; Generated by FilamentChange plugin
;LAYER:19
;TYPE:WALL-INNER
;MESH:xyzHollowCalibrationCube.stl
G1 F1500 X126.5 Y115.471 E419.54431
G1 X124.5 Y115.471 E419.61016

Ghostkeeper, 2021-04-08:
@Wob76 maybe there is a difference in the Retract Before Layer Change setting?

Wob76, 2021-04-08:
@Ghostkeeper I have that off by default, and I kept all Cura settings the same in the above examples. all I did is remove the original STL and place a different STL (a test cube), so it just seems to be related to how Cura is planning an object, in this case the first is way more complicated than the later, but I am not sure if complexity is part of it or not. I probably need to just do some testing with various STL files and see which ones have the extra retracts in place.

Ghostkeeper, 2021-04-15:
Without a project file there's hardly anything I can experiment to see what is the issue.

Perhaps it's just the retraction limit? What happens when you increase the Maximum Retraction Count setting considerably (like, add a 0 to the end)?

jmerc77, 2021-04-16:
I think i found something...
no mater settings, if retraction is on it always retracts on layer change. this is with version 4.8.0

jmerc77, 2021-04-16:
my settings/model that caused this:
[CCR10S_digits.zip](https://github.com/Ultimaker/Cura/files/6327993/CCR10S_digits.zip)

Ghostkeeper, 2021-05-04:
> no mater settings, if retraction is on it always retracts on layer change. this is with version 4.8.0

With this model, that is expected behaviour. That's because as it goes to the next layer, it also travels to the part that is closest to the Layer Start X and Layer Start Y position. This travel move goes across the walls to a different part, so according to the normal travel move routine it should then retract. The entire travel move is retracted; it's not going to unretract specifically for the Z move and then retract again for any horizontal travel moves it needs to make or anything.