Ultimaker/Cura #8950: 4.8.0 - 3MF Overrides Not Accurately Labeled

PriddyPrints, 2020-12-15:
**Application version**
4.8.0

**Platform**
Mac OS 10.15.7

**Printer**
S5

**Reproduction steps**
Any time a 3MF is loaded once the behavior begins to occur. 

**Screenshot(s)**
[image]

**Actual results**
Works as expected after a Cura reset, but will occur again a short time after. Based on the behavior so far, it begins mimicking the override count from a previously loaded 3MF onto all future 3MFs that are loaded. The correct overrides are loaded when viewing the custom settings tabs, so it looks to be purely a visual bug and not impacting actual configurations. Also not able to reproduce this on other colleague's Cura 4.8 installations on Mac OS.

**Expected results**
Should show the correct amount of overrides in the 3MF. For example, the provided file should have close to 5.

**Project file**
[Test.3mf.zip](https://github.com/Ultimaker/Cura/files/5698553/Test.3mf.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5698557/cura.log)

If anything else is needed, please let me know.

konskarm, 2020-12-17:
I don't think we have touched that code for a long while. Can you test if it happens to you also on previous versions?

konskarm, 2020-12-17:
I was able to reproduce it after all in the current master branch. I'll discuss with the team.

PriddyPrints, 2020-12-17:
Awesome! I appreciate you looking into it.

Ghostkeeper, 2020-12-22:
This is to be put in our backlog. (@Vandrasc don't forget!)

closed 2024-12-08 by github-actions[bot]