Ultimaker/Cura #8952: Windows 10 incorrect controls placement with 150% wide screen setting.

larivcl24, 2020-12-16:
**Application version**
4.8.0

**Platform**
Windows 10
**Printer**
FLSun QQ-S pro

**Reproduction steps**
1. print
2. monitor

**Screenshot(s)**
[image]

**Actual results**
Can't see actual progress.

**Expected results**
Monitor the print...

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-12-16:
None of these out of place controls are provided by Cura itself; they are added by a plugin. Please find the issue queue for the plugin and report this over there instead.

larivcl24, 2020-12-16:
Ok

closed 2020-12-16 by nallath