Ultimaker/Cura #8956: Switch printer will change print settings

louis1lal, 2020-12-16:
Application version
4.8.0

Platform
OSX 10.13.6 (High Sierra)

Printer
Anycubic Mega S

Reproduction steps

1. Open Cura
2. Open a project with a different printer
3. Switch to an other printer

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
If the printer uses different print settings by default, the current print settings will switch to the one used by the printer selected

**Expected results**
If I open a project, the goal is to use the exact same print settings but sometime I only want to switch printer. So changing printer should not change print settings.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-12-16:
I'm not quite sure what you mean.

If you want to switch printer, please use the machine selector. If you don't want the machine settings from a project file (but only the models) use the "load models only" option. 

Could you elaborate on what you mean?

louis1lal, 2020-12-17:
Maybe a video would be more clear I can share a screen cast link from my google drive. Let me know if you want me to do it.

But,
Lets say I want to create an other project with an existing one, but only to use it with an other printer. So I want all settings the same but I want to change the printer. If I do that, for some reason, the print settings will change. It's not what I want.

Sorry it's kind of hard to explain better than this, otherwise I'll screen cast my screen.

nallath, 2020-12-18:
I think sharing a video would help.

louis1lal, 2020-12-18:
OK I'll do it as soon as I have some time so do not close this issue if I do not reply that fast

Thanks

Ghostkeeper, 2020-12-22:
You can select which printer is used from a drop-down when it opens the project file, right? At least, from among all of the printers that have the same printer model.

closed 2020-12-24 by louis1lal