Ultimaker/Cura #8957: Weight variance

CheoRamos, 2020-12-16:
**Application version**
4.7.0

**Platform**
Microsoft Windows 10 Pro

**Printer**
Ultimaker 2+

**Reproduction steps**
1.  CURA is estimating 442 g for a piece and I'm obtaining 328 g
in the printed one.
2.  The density of the filament is correct and I verified I'm entering the same
infill percent.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
What else should I do to improve this estimate?

jmding8, 2020-12-17:
Have you calibrated your extruder? https://www.youtube.com/watch?v=YUPfBJz3I6Y&ab_channel=ThomasSanladerer

Ghostkeeper, 2020-12-22:
This means you're either experiencing underextrusion, or the density in the material profile is not correct.

To reduce underextrusion, I suggest you take a look at the troubleshooting article in the Settings Guide which has quite a few suggestions: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/underextrusion.md

The material density can be edited for custom materials in the material manager:
[image]
If you're using a specific material which is not as dense as the others, this is the correct way to calibrate it.

To consider this a bug, we'd need to know a bit more about your problem. For instance, could you provide a project file?
Using the project file we can verify the actual amount of material that Cura is instructing your printer to extrude, to see whether its estimate is correctly aligned with the g-code file. We could then rule out that it's just a print quality issue.

closed 2021-01-05 by no-response[bot]