Ultimaker/Cura #8962: [4.8] The settings from the material are not applied to the profile

userosos, 2020-12-17:
Hi! If i changed settings at profile after it settings from Material not applay. I tested it with Material Settings plugin and without one. It not work. It looks like bug.
If it is not bug can you add checkbox into Material "Make the material settings master" and settings from Material rewrite Profile?

**Application version**
Cura 4.8.0

**Platform**
Windows 10 2020 x64, Quadro P1000 (laptop)

**Reproduction steps**
1. Change an settings into profile. Temp bed, for example
2. Change the settings into material. Temp bed, for example
3. Settings from material don't Apply.
4. If i use Material Settings plugin i can't use rewrite all of them settings Which I've changed before in profile.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

**Project file**
[ult02_separator_v1_ULTIA.zip](https://github.com/Ultimaker/Cura/files/5710523/ult02_separator_v1_ULTIA.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5710532/cura.log)

konskarm, 2020-12-21:
I am closing this as a duplicate of https://github.com/Ultimaker/Cura/issues/7626. Please post any additional comments that you have at that issue.

closed 2020-12-21 by konskarm

Ghostkeeper, 2020-12-22:
The problem here is actually different. That issue is about the profile manager which gets its list of profile overrides with a ListModel that is incorrectly filled for the extruders. This settings list just shows all settings. In this case it's a settings list created by the MaterialSettings plug-in too.

However the actual problem is that the user's custom profile "Ulti ABS_normal" overrides the temperature and as such the material's temperature is not used any more. It's explicitly overridden by the custom profile. That's working as expected.